riverloopsec/killerbee #34: zbassocflood not working

junkmailsucksreallybad, 2015-05-19:
trying to run zbassocflood -h and I get ": No such file or directory"

Tried on Debian 32bit, Ubuntu 64bit, and Archlinux - all same error.  Is this a known good tool?  I did not get Scapy properly installed yet but it doesn't seem this is required?

joswr1ght, 2015-05-19:
Is zbassocflood in your PATH somewhere?  What does `which zbassocflood` return?

If it doesn't return anything, make sure you ran "sudo python setup.py install" to install KillerBee properly.  By default, this will install in the Python local directory (e.g. /usr/bin).

junkmailsucksreallybad, 2015-05-19:
yes it is - all other zb* tools work fine.  

It returns:
>>zbassocflood -h
": No such file or directory"

Crypt0s, 2015-06-04:
I ran into this problem, specify "python zbassocflood" and it will work.

riverloopsec, 2015-06-29:
All the scripts have the same #!/usr/bin/env python header and are installed by setup.py in a long scripts array.
@junkmailsucksreallybad - did @Crypt0s' suggestion help in your case? Also your output seems to show 'which zbassoctool' - I'm assuming this was a typo when reproducing your output here?
@Crypt0s - if running it with python is needed for you, can you check the output of 'which zbassocflood' as well?

junkmailsucksreallybad, 2015-06-30:
Yes @Crypt0s suggestion did make things work.  It was confusing why the others worked but this one did not.  Maybe something was missing from the install script or something happened to break at that moment, but this one specifically didn't work on 3 different machines I tried. 

'which zbassocflood' doesn't give me back anything.

riverloopsec, 2015-07-05:
Ok. Can you look in your setup.py script and make sure the line that installs the scripts has this in it, like this:
```
        scripts = ['tools/zbdump', 'tools/zbgoodfind', 'tools/zbid', 'tools/zbreplay',
                   'tools/zbconvert', 'tools/zbdsniff', 'tools/zbstumbler', 'tools/zbassocflood',
                   'tools/zbfind', 'tools/zbscapy', 'tools/zbwireshark', 'tools/zbkey',
                    'tools/zbwardrive', 'tools/zbopenear'],
```
If that is the case, maybe try rerunning sudo python setup.py install again and see if that addresses it. On success, you should be able to do:
```
$ which zbassocflood
/usr/local/bin/zbassocflood
```
If that isn't the case, maybe the logs from setup.py could help point at the issue?

rmspeers, 2016-11-05:
I am closing with the presumption that this is resolved. Please let me know if not.

closed 2016-11-05 by rmspeers