riverloopsec/killerbee #35: Aligning attack to existing Beacon TDMA frames

junkmailsucksreallybad, 2015-06-04:
Is there any existing framework (either KillerBee or the base AVR Firmware) to build on which listens for existing Beacons on a specified channel, and uses that beacon to align the requests made by KillerBee?  For example, using zbassocflood on channel 11, would it be more effective in a beacon-enabled network to try and transmit in all timeslots or at the very least specific timeslots?

I am just wondering how effective the transmissions are if they are not aligning to timeslot boundaries.  I may be misunderstanding a portion of the protocol but this seems like an important factor.

Crypt0s, 2015-06-04:
Someone can correct me if I'm wrong but:

The only portions of the 802.15.4/Zigbee spec that specify timeslots are the GTS slots in beacon-enabled networks, which must be negotiated with the PAN coordinator.  Otherwise, transmissions are on a Carrier Sense Multiple Access / Collision Avoidance  (CSMA/CA) deal.  Only a few (8, i think?) GTS slots are available (and only on beacon-enabled networks, where timing information can be sync'd)

riverloopsec, 2015-06-29:
@junkmailsucksreallybad - does the network you are looking at have GTS extensions in use when you look at the raw beacons (ex zbwireshark)?

Also, my _first thought_ is that if you're sending frames to the coordinator like in zbassocflood, it should be listening all the time and not just for scheduled frames. However, this could also be dependent on receiving chip or firmware.

junkmailsucksreallybad, 2015-06-30:
This was just my mistake in understanding the protocol.  GTS is not enabled but I was not seeing any effect by end devices with zbassocflood.

taylorcenters, 2021-06-22:
closed - long idle

closed 2021-06-22 by taylorcenters