riverloopsec/killerbee #36: Makefile for RZUSBSTICK firmware is Windows only

cybergibbons, 2015-06-23:
The currently Makefile is for Windows only. It should be possible to make this work for both Windows and Linux.

https://github.com/riverloopsec/killerbee/blob/master/firmware/src/kb-rzusbstick/gcc/RZUSBSTICK/Makefile

riverloopsec, 2015-07-05:
@cybergibbons  - please see if the new update to firmware/src/kb-rzusbstick/gcc/Makefile works for you on Linux. If so can you close this issue?

Scytmo, 2017-05-22:
Works for me if I remove --gc-sections.  See my related issue #90

rmspeers, 2017-07-04:
Given the info in #90  that @Scytmo  has provided, I am fine with an update to remove that option and merge this all down to master. I will plan to do that linked to that ticket. Given this ticket has been open for a while without updates, I'm going to close it out as it appears to be working.

closed 2017-07-04 by rmspeers