riverloopsec/killerbee #38: kbdecrypt doesn't correctly determine data/MIC boundary (possibly scapy-com issue)

cybergibbons, 2015-06-23:
These lines of code in scapy_extensions.py:

    pkt = pkt.copy()   #this is hack to fix the below line
    pkt.nwk_seclevel=5 #the issue appears to be when this is set

    #mic = struct.unpack(">I", f['mic'])
    mic = pkt.mic

The mic is always blank in this case. I suspect this is because when the packet is parsed initially, the mic length is set to 0. Changing the nwk_seclevel after this has no impact.
```
def util_mic_len(pkt):
    ''' Calculate the length of the attribute value field '''
    if ( pkt.nwk_seclevel == 0 ): # no encryption, no mic
        return 0
    elif ( pkt.nwk_seclevel == 1 ): # MIC-32
        return 4
    elif ( pkt.nwk_seclevel == 2 ): # MIC-64
        return 8
    elif ( pkt.nwk_seclevel == 3 ): # MIC-128
        return 16
    elif ( pkt.nwk_seclevel == 4 ): # ENC
        return 0
    elif ( pkt.nwk_seclevel == 5 ): # ENC-MIC-32
        return 4
    elif ( pkt.nwk_seclevel == 6 ): # ENC-MIC-64
        return 8
    elif ( pkt.nwk_seclevel == 7 ): # ENC-MIC-128
        return 16
    else:
        return 0
```
Not sure if this is scrictly a killerbee issue.

riverloopsec, 2015-06-29:
Agreed that ENC-MIC-32 is the most common by far. Are you able to make a small script to reproduce this issue just using scapy-com parsing? If so we can focus on addressing that. The workaround lines you posted from scapy_extensions were contributed, so we're not familiar with why that workaround is needed but if we can find the error is with how scapy-com is being used, we can see if that needs changing as well.

cybergibbons, 2015-06-29:
I'm about to ask on stack-overflow about it, as I don't quite understand how scapy is meant to work in these situations.

onizenso, 2015-07-08:
If I read from a pcap file using `kbrdpcap(file_name)`, I get the errors about the FCS checksum not being there. Removing the FCS checksum (mic), I passed the parameter `nofcs=True` to `kbrdpcap(file_name, nofcs=True)`. No errors after that, but there are checksums in the sample I have. Could Killerbee's parser be responsible, maybe scapy-com parses differently? 

Not sure that directly relates to your issue, but I noticed a similar problem in the sister-thread on Google Code.

superdupertestuseroutlook, 2016-04-27:
I am seeing the same issue as well that MIC is always empty - I don't see errors about no FCS, I'm getting stuck prior to that, as listed in my new post regarding the error: "struct.error: integer out of range for 'L' format code"

rmspeers, 2016-04-28:
Do you have an example script and PCAP that you can upload or provide privately for me to look at this on?

Scytmo, 2017-04-11:
I have fixed this in my pull request #89 and also fixed the MIC calculation such that kbdecrypt can now be used to test whether a NWK key is correct.  Please see my detailed commit message in that pull request for details of the issue and my proposed fix.

closed 2021-06-22 by taylorcenters