riverloopsec/killerbee #39: zbdump memory corruption

ridingintraffic, 2015-06-29:
i have been receiving this error a lot today.   is it a flaw in the firmware i built or a software issue?

zbdump: listening on '001:005', link-type DLT_IEEE802_15_4, capture size 127 bytes
*** glibc detected *** /usr/bin/python: malloc(): memory corruption: 0x010a29b0 ***

riverloopsec, 2015-07-05:
It looks like your python process itself has memory corruption... which seems very odd. Did you recently upgrade to a new python program? Do you get it when running other tools that capture data, like zbwireshark? What about when sending frames (zbreplay)? What about on non KillerBee python programs?

rmspeers, 2015-09-16:
Closing but let me know if we need to reopen.

closed 2015-09-16 by rmspeers