riverloopsec/killerbee #40: zbid fails with 'module does not contain serialutil'

onizenso, 2015-06-29:
After a fresh install of Arch Linux with the BlackArch user repository, the KillerBee framework fails to install properly. Even with all of its dependencies met, zbid fails to launch, producing the error 

```
           Dev Product String       Serial Number
Traceback (most recent call last):
  File "zbid", line 23, in <module>
    show_dev(gps=arg_gpsdev, include=args.include)
  File "build/bdist.linux-armv7l/egg/killerbee/__init__.py", line 45, in show_dev
  File "build/bdist.linux-armv7l/egg/killerbee/kbutils.py", line 223, in devlist
  File "build/bdist.linux-armv7l/egg/killerbee/kbutils.py", line 289, in isgoodfetccspi
AttributeError: 'module' object has no attribute 'serialutil'
```
Any help, or guiding hints are very welcome. Has anyone else on Arch run into similar problems?

P.s. I tried installing from the package in the BlackArch repo and from the most recent, master build on github. I am also installing to a BeagleBone Black Revision C, running a Cortex A8 processor.

Thanks in advance for any help.

riverloopsec, 2015-07-05:
Interesting - it looks like your Linux install is either causing a path to trigger that usually doesn't, or the python serial library you have doesn't contain something we expect to be there.

The code hitting this issue is in killerbee/kbutils.py:
```
    try:
        gf.serInit(port=serialdev, attemptlimit=2)
    except serial.serialutil.SerialException as e:
        raise KBInterfaceError("Serial issue in kbutils.isgoodfetccspi: %s." % e)
```

Can you please do the following to debug:
1) try opening a python interactive shell, type `import serial` and then try to access `serial.serialutil` and `serial.serialutil.SerialException`. See what the output there is.
2) in the code snippet above, replace the `serial.serialutil.SerialException` with `Exception`. You may need to change two other instances of this in the same file (let me know if so). Then provide the output from a zbid run that hits the exception.

onizenso, 2015-07-06:
Thanks @riverloopsec, I think it might have been tied to the serial.serialutil.SerialException. I just ran zbid again on my BeagleBone Black running BlackArch, and it seems to be running fine. Don't know exactly what fixed it, but I will keep you updated if I find it.

closed 2015-07-06 by onizenso