riverloopsec/killerbee #42: zbassociation : Association header  byte error

EastL, 2015-07-02:
Hello ~ When I use the zbassociation function I found that some bytes are wrong.
https://github.com/riverloopsec/killerbee/blob/master/tools/zbassocflood#L77

The last 2 bytes should be 0x01(Command Identifier) and 0x8e(Association Request), and when I fixed it I get association response sucessfully.

riverloopsec, 2015-07-05:
So are you saying the 0x67 should be removed from that byte string? If so, that makes sense -- I'm not sure why that is there to start. This code may need some reworking in general after I took a look at it just now. If I do a more heavy-handed cleanup of the code, would you be willing to test?

riverloopsec, 2015-07-05:
Did you also have to remove the 0x67 byte in this code to get it to work for you?
```
        ffdswitch = not ffdswitch
        if ffdswitch:
            assocreqp[6] = "\x01\x8e\x67"  # Full Function Device (FFD)
        else:
            assocreqp[6] = "\x01\x8c\x67"  # Reduced Function Device (RFD)
```
What ZigBee version/spec does your target implement? Is it a full or reduced functionality device?

rmspeers, 2016-04-30:
@EastL - are you still able to reproduce this and can help me understand the suggestion you made per the questions above?

taylorcenters, 2021-06-22:
closed - long idle

closed 2021-06-22 by taylorcenters