riverloopsec/killerbee #52: Question: Freakduino Support

bneg, 2015-08-27:
The documentation says support is in development - has there been any activity on this? Thanks!

rmspeers, 2015-09-16:
There was activity, and we did support and use it in research efforts for a time. However, it was based on an Arduino sketch running on the Freakduino (initial version) and only supported sniffing. The code is available, but at least the main project members haven't used it in a few years. I'd love to hear if you have interest and if it works for you (it should still). Let me know if you need pointers to any code, etc. 
The device handler is at: https://github.com/riverloopsec/killerbee/blob/master/killerbee/dev_freakduino.py
You need to enable it in:
https://github.com/riverloopsec/killerbee/blob/master/killerbee/config.py
The code for the sketch is at:
https://code.google.com/p/zigbee-security/source/browse/#svn%2Ftrunk%2Fkillerbee_extras_platforms%2Ffreakduino%2Ffirmware
Because the last code is on Google Code still, if it works for you, please pull the code and create repository to this repo, storing it under the firmware/src/freakduino directory.
Much appreciated!

rmspeers, 2016-01-11:
Please reopen if you have additional questions or need help with this!

closed 2016-01-11 by rmspeers