riverloopsec/killerbee #55: zbstumbler: Exception: Error: Semantical Error

ad0nis, 2015-10-01:
Whenever I run zbstumbler, after some amount of time (usually after a few networks are spotted, but not always,) I get the following error:

Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 5, in <module>
    pkg_resources.run_script('killerbee==2.5.0', 'zbstumbler')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1401, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.5.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 191, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 292, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 438, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 334, in sniffer_on
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 306, in _open_stream
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 267, in __usb_write
Exception: Error: Semantical Error

rmspeers, 2016-04-30:
Some people have reported similar issues, apparently something in that RZUSBSTICK firmware on glib or something. One person patched by restarting the sniffer every 25 frames, another uses a different method. We could add those into zbstumbler on a branch if you're still able to reproduce this, and then see if that resolves it for you?

Crypt0s, 2016-05-03:
I've noticed this behavior stems from switching contexts on the RZUSBStick from sniff to inject a few times, a behavior present in zbstumbler iirc.

re: the 25 packet fix:
I recall tracking this down to a firmware issue on the stick, which is why the fix I implemented after 25 attempts at reading the packet resets the usb interface and re-loads the RZUSBStick reference.

thecheshirecat15, 2018-05-02:
I have exactly the same exception, with every firmware version on the original rzusbstick.
It happens when I use zbassocflood or when I use directly the framework in my code.
I haven't found the 25 packet patch, is it normal ? How could I solve this problem ?

Pchewing17, 2018-09-05:
any update on the status of this fix?

Warning: You are using pyUSB 1.x, support is in beta.
zbstumbler: Transmitting and receiving on interface '1:5'
Transmitting beacon request.
Traceback (most recent call last):
  File "/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbstumbler')
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 661, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/lib64/python2.7/site-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 191, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 357, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 546, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 385, in sniffer_on
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 355, in _open_stream
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 334, in __usb_write
Exception: Error: Semantical Error

rmspeers, 2018-09-06:
@thecheshirecat15 the reloading the interface seems like what @Crypt0s  had put within the code in some places. I'm wondering if there is a code path that doesn't benefit from that internal patch or something. Do you only get this when you try to inject, instead of just sniffing (like zbassocflood and zbstumbler vs zbdump and zbwireshark)? Or is it a tool by tool issue regardless of tx vs rx?

closed 2021-06-22 by taylorcenters