riverloopsec/killerbee #60: KillerBee on Pentoo Linux with Sewio Open Sniffer

tahersb, 2016-02-10:
Current Setup:
OS: Pentoo Linux 2015.0 RC4.6 image booted from USB stick
Hardware: Sewio Open Sniffer v2.0 Firmware version: 0.8.5 connected to host on eth0

Sanity tests:
- OpenSniffer web-app accessible on 10.10.10.2
- KillerBee and all required dependencies including scapy successfully installed on Pentoo. Commands like zbid -h work well.

Problem:
zbid does not recognize OpenSniffer. No dev string is displayed. 
zbstumbler --iface eth0 gives error indicating OpenSniffer hardware not detected.

Can someone please help me understand if OpenSniffer v2.0 is actually supported by KillerBee? What can I do to fix this?

ran2, 2016-02-13:
I also wonder how to make KillerBee work with Sewio Open Sniffer because Open Sniffer connecting thro' eth0 but Atmel's AVR RZUSBSTICK connecting thro' usb port ... !

Crypt0s, 2016-02-17:
After reading the code, try the following:

Hook up the sniffer to the thernet port + power it
Then, type:
`zbid --dev 10.10.10.2`
Replace 10.10.10.2 with your custom IP address if you changed it from the default.  My sniffer seemed to balk at this, but not sure if it was because network manager was getting in the way of the interface IP settings.

Lemme know if that helps.

rmspeers, 2016-04-03:
Did that help resolve your issue?

rmspeers, 2017-06-25:
@tahersb , @ran2  - Did the above suggestion work for you? I don't have a sniffer any more as mine got lost sharing it with someone, so I can't test specifics on this.

rmspeers, 2018-06-11:
@tahersb , @ran2  - Closing as I believe this is resolved by a more recent PR/ticket if the above didn't work for you. Please reopen if needed.

closed 2018-06-11 by rmspeers