riverloopsec/killerbee #61: kbdecrypt complains about source addr out of range

superdupertestuseroutlook, 2016-04-27:
When I try and decrypt a packet I see:

Connected to pydev debugger (build 145.844)
    Packet 0
        Dot15d4FCS
        Dot15d4Data
        ZigbeeNWK
        ZigbeeSecurityHeader
        Unknown Encrypted App Layer
Traceback (most recent call last):
  File "/home/testuser/pycharm-community-2016.1.2/helpers/pydev/pydevd.py", line 1531, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "/home/testuser/pycharm-community-2016.1.2/helpers/pydev/pydevd.py", line 938, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/testuser/Desktop/zigbee_tools-master/LAYER_identifier.py", line 148, in <module>
    enc_data = kbdecrypt(data[e],network_key)
  File "/usr/local/lib/python2.7/dist-packages/killerbee/scapy_extensions.py", line 442, in kbdecrypt
    nonce = struct.pack('L',f['source'])+struct.pack('I',f['fc']) + sec_ctrl_byte
struct.error: integer out of range for 'L' format code

debugging down to scapy_extensions.py further shows something like this when looking at the layer fields (I added FF instead of real data in some places - this is "f" array just before nonce is computed line 442  :

{'key_seqnum': 0, 'nwk_seclevel': 5, 'reserved1': 0, 'key_type': 1, 'source': 3123456789123456L, 'fc': 2111111, 'extended_nonce': 1, 'data': '6\x0F\x8F\xFF\xFF\x84^\r\x935&\xFF\xFF\xFF\xFF\xFF\xFF\xFF'}

It seems like the "Source" is very large integer instead of the formatted 00:d0:10:00:00:00 etc....I know the key is correct and the pcap is valid because wireshark decodes this properly, network   Any ideas?

superdupertestuseroutlook, 2016-04-29:
This is fixed by adding a 'Q' instead of 'L' for:File "/usr/local/lib/python2.7/dist-packages/killerbee/scapy_extensions.py", line 433

Didn't work:
nonce = struct.pack('L',f['source'])+struct.pack('I',f['fc']) + sec_ctrl_byte

Works:
nonce = struct.pack('Q',f['source'])+struct.pack('I',f['fc']) + sec_ctrl_byte

Obviously this may be dependent on my capture file vs what anyone else is testing, so use only as needed I guess.

rmspeers, 2016-04-30:
Is it worth reopening this and making the change on a branch, and then we can see if others find that change breaking or helpful for them as well?
Actually, maybe the default could stay as 'L', but if there is a way to switch to 'Q' based on checking something about f['source'], then maybe it could switch to Q?

superdupertestuseroutlook, 2016-05-02:
I only have pcaps available right now which have the extended address.  If you send me one which works fine with 'L' I can certainly try and fix that up.  its possible it may work with Q as default otherwise we can do the if statement to sort it out based on some other bits - we can probably see how the wireshark filter knows how to differentiate and mimic that.

Clark

taylorcenters, 2021-06-22:
closed - long idle

closed 2021-06-22 by taylorcenters