riverloopsec/killerbee #62: kbencrypt issue - not finding 'reserved2'

superdupertestuseroutlook, 2016-04-29:
kbdecrypt seems to work fine - however when I try and re-encrypt using kbencrypt (the unencrypted blob just decrypted) I am getting some errors saying that 'reserved2' is not defined:

 File "/usr/local/lib/python2.7/dist-packages/killerbee/scapy_extensions.py", line 510, in kbencrypt
    fc = (f['reserved1'] << 6) | (f['extended_nonce'] << 5) | (f['key_type'] << 3) | f['reserved2']
KeyError: 'reserved2'

Any known issue in the area?  I haven't seen anything posted regarding this.

superdupertestuseroutlook, 2016-04-29:
Can anyone explain what the bold line actually trying to accomplish in dot15d4.py?

class ZigbeeSecurityHeader(Packet):
    name = "Zigbee Security Header"
    fields_desc = [ 
        # Security control (1 octet)
        **HiddenField(FlagsField("reserved1", 0, 2, [ 'reserved1', 'reserved2' ]), True),**

It also appears 'reserved2' was commented out in kbdecrypt for some reason.

superdupertestuseroutlook, 2016-04-29:
Just want to make sure my error isn't a result of me using the API wrong - Can anyone give a proper use for kbencrypt(pkt, enc, network_key) - 

Can **pkt** represent a normal packet right out of a pcap (with encrypted payload) and **enc** being the decrypted payload from that same packet, decrypted with kbdecrypt?  

I just wanted to loop through a pcap file, decrypt each packet/payload, print it, then use kbencrypt to re-encrypt the payload and verify I can produce the identical output pcap as the original, to ensure my KB framework is all working properly.

rmspeers, 2016-04-30:
I have posted a Tweet to the author of this module to ask for his insight here on what that line means, what some of the issues may be, etc. https://twitter.com/rmspeers/status/726504996888547328

zeroSteiner, 2016-04-30:
I'll take a look at this.

zeroSteiner, 2016-04-30:
Alright, I've found the issue. As it turns out, [this commit](https://bitbucket.org/secdev/scapy-com/diff/scapy/layers/dot15d4.py?diff2=437f5fe62c32&at=default) introduced in the upstream scapy-com repository removed the `reserved2` field of the `ZigbeeSecurityHeader`.

It looks like it can simply be renamed to it's new value of `nwk_seclevel` to fix this bug. I'll submit a PR here in a minute.

rmspeers, 2016-04-30:
Thanks @zeroSteiner  for the quick fix.
@superdupertestuseroutlook - can you go to his branch and confirm this fixes it for you and let us know?

superdupertestuseroutlook, 2016-05-02:
Yes this fixed the issue - I simply made the changes locally and it works fine now, I can decrypt and re-encrypt just fine!  Wooohooo!

Thanks for such quick responses!

Clark

commit 245fa2ba (2016-05-06): Merge pull request #63 from zeroSteiner/fix-62

Fix #62, update reserved2 to nwk_seclevel
commit e2e2c5d5 (2016-05-06): Fix #62, update reserved2 to nwk_seclevel

Update the ZigbeeSecurityHeader.reserved2 field to
nwk_sec_level to reflect changes introduced in the
upstream scapy-com repository.
closed 2016-05-06 by rmspeers