riverloopsec/killerbee #65: kbencrypt issue - adding MIC incorrectly?

superdupertestuseroutlook, 2016-05-04:
I am attempting to simply test the framework by reading in a packet from a pcap which has encrypted traffic, decrypting it and re-encrypting it and sending it back out.  I am trying:

..reading in pcap...
for e in range(num_pkts):
   decrypted_data = kbdecrypt(data[e],network_key)
   re_enc_data = kbencrypt(data[e], decrypted_data , network_key)

Shouldn't data[e] = re_enc_data?  It appears after calling kbencrypt, the MIC is placed onto the packet in a goofy way at the end of the packet.  Am I calling the function incorrectly?

superdupertestuseroutlook, 2016-05-04:
I found the issues I believe - I have a handful of edits to scapy_extenstions.py both to mimic the pkt.copy bug cutaway fixed in zbdecrypt which needed to be in the encrypt as well, and also simple commented out adding the MIC second to last line in kbencrypt.  Not 100% certain why that works yet but still need more verification.

Would you like the code pasted here or to pull a branch?  

Does anyone have experience testing something with kbencrypt so they would know if my changes helped or made things worse?  I have no idea how anyone was using that function before and had it work.

superdupertestuseroutlook, 2016-05-05:
With more verification, I can see things were only fully working because i was just spitting out the same data as I was pulling in, and commenting out the MIC replacement meant the original MIC was correct and things looked good.  

However, after trying to actually modify some of the plaintext data (decrypted_data as seen above) and re-encrypt it I noticed wireshark couldn't decrypt my data anymore, so it appeared removing teh MIC was bad.  I added this back in, and modified the code to properly replace that field not tack it on the end, but I still have no luck.

Has anyone verified that the MIC works as part kbencrypt????  I could really use some help.

superdupertestuseroutlook, 2016-05-05:
I made my changes to a branch I created from master, I can't seem to push my branch though I get a 403 failed.  Do I need to request permission somehow to get access to posting a branch on here?

rmspeers, 2016-05-06:
Usually what people to do for GitHub PRs is to create a fork of the repository they own, push the branch there, and then submit a PR to the main repository. Would that work? I'm also happy to manually update them if you want to post the files, but if you already have the changes a PR is great.

I have not verified that the MIC works in kbencrypt. My best idea for an easy test it is to generate some frames, save to PCAP, and then load the key in Wireshark and test that it is showing the MIC as valid. I'm happy to try this with you if needed.

superdupertestuseroutlook, 2016-05-06:
Your suggestion for an easy test is exactly what I am doing - I originally thought I fixed in my super simple test where I commented out setting the MIC (because original code appends data it seems in ascii) - and it was working when I just re-encrypted the same thing I read in (because MIC was being left alone).  Now I tried setting the MIC properly, and I can see it is not being computed correctly because wireshark stops decrypting properly when I do set the MIC.

Do you have some PCAP / Network key you can work with?  I cannot give you a PCAP and a key I am using specifically.  I would guess there is some example PCAPs out there using the default key or something we could both test with?

I checked in my changes to https://github.com/superdupertestuseroutlook/killerbee  - PR on the way.

rmspeers, 2016-05-08:
I've carved out a packet and it's key from a capture I can share. Please email me at ryan at riverloopsecurity and I'll send you that PCAP and key to use as a test. It is an AES-128 and 32-bit MIC frame if that works. I have two of them if needed.

superdupertestuseroutlook, 2016-05-09:
I have it working now - with changes made to scapy_extensions.py and I checked in my test script under killerbee/scripts/testKBencrypt.py all to my branch, and created a PR.  I probably need to clean up the script to pull off the MIC code in "decrypted bytes" comparison, but everything else works.

I can verify now that in Wireshark I get the same decrypted bytes after decrypting the original pack, and my rebuilt packet from the decrypted data.

I'll let you verify/merge and officially close the bug - but to me the bug is fixed.

closed 2021-06-22 by taylorcenters