riverloopsec/killerbee #67: Real-Time Listen And Processing of ZigBee Packets?

superdupertestuseroutlook, 2016-05-10:
Is there is a way to do real time capture and processing of the data?  I think I need to use a proper sequence number for fuzzing some ZigBee stuff, and I assume I need to listen to broadcast packets and understand the proper sequence number to achieve this - currently sending valid commands and fuzzing blindly doesn't work even though I have proper key for encrypted commands.

Any help / suggestions?  This seems like a fast enough protocol maybe that has to be developed in firmware, but could be something added to killerbee firmware to listen to broadcast frames so there could be a kbGetNextSeqNum() exposed in software - or something of that order?  Maybe this would require two KillerBee devices but it seems like it could be done.

Otherwise it seems like sending commands and fuzzing is basically useless without this functionality on a system that implements this properly.

rfmcpherson, 2016-05-10:
This looks similar to what the wardriving code does. You might be able to patch your functionality there. I have a PR that cleans up the wardriving somewhat. I'm hoping to clean it up a little more and get it merged after the CCS deadline on the 23rd.

rmspeers, 2016-06-11:
Yes, the wardriving code may help -- I previously had code for this (not in the repo) that kept a database of the network state based on passive sniffing, and then allowed you to issue injection commands that filled in the right data based on current network state. On most networks it was fast enough, on others it's not going to work -- even with firmware tracking and exposing to software the next number may have issues on reallyyyy busy networks (luckily most are not so should be fine).

closed 2021-06-22 by taylorcenters