riverloopsec/killerbee #68: zbgoodfind - can't find encrypted packets?

superdupertestuseroutlook, 2016-05-16:
I am attempting to run zbgoodfind with a Zigbee PCAP which I am 100% sure has encrypted packets (I enter the key in order to see the payload unencrypted in wireshark).  Is is possible zbgoodfind is not looking at low enough layers, or is it checking only for encryption at the highest layers?

rmspeers, 2016-05-21:
What are you feeding in for the firmware dump argument?

superdupertestuseroutlook, 2016-05-23:
zbgoodfind  -f Desktop/firmwareFiles/firm.bin -r Desktop/turnItOff.pcap
...
No encrypted packets found in the capture file Desktop/turnItOff.pcap.

The PCAP file definitely has encrypted packets using the network key, and the .bin file is simply an ELF format Linux firmware image.

rmspeers, 2016-05-24:
I assume you don't know the key via some other means (e.g. from OTA pairing, etc). If so, that could be used to debug and determine why zbgoodfind didn't retrieve it. The logic in zbgoodfind is pretty basic, and could fail from either (off the top of my head) the key not being in order in the firmware file, or the decryption attempts on the packets in the PCAP being incorrect.

superdupertestuseroutlook, 2016-05-26:
Yes I do know the key - I just wanted to see if this tool could find it in a firmware image.  

The error message saying specifically the PCAP file has no encrypted packets, would be a bit odd if that meant that it couldn't find a key.

Unfortunately I can't provide a key and firmware image for debug, but is there one available to ensure its not my test tools?

joswr1ght, 2016-05-26:
Sorry, I should have chimed in sooner. Zbgoodfind only handled 802.15.4 encryption today, since I wrote it before we had full NWK and APS encryption support in KillerBee.

I think it should be straightforward to extend Zbgoodfind to also handle NWK and APS encrypted packets as well...

-Josh

superdupertestuseroutlook, 2016-05-26:
Ok - cool, thanks.  I will try to and extend it when I get some time, others if you try this please update this thread or let me know.

SteveJM, 2018-12-19:
I expect to build on my zbsniff changes to look at this tool as well.

closed 2021-06-22 by taylorcenters

DuckSounds, 2023-04-20:
Were there any changes to this?