riverloopsec/killerbee #70: `pnext` times out on OSX 10.11.5

sxmichaels, 2016-06-27:
I am attempting to use the latest Killerbee on OSX 10.11.5. I am using Python 2.7.1, with pip packages  at the following versions (I don't know exactly which are KB dependencies so here's the whole list):

```text
crypto (1.4.1)
futures (3.0.5)
greenlet (0.4.10)
jedi (0.9.0)
killerbee (2.6.0)
msgpack-python (0.4.7)
Naked (0.1.31)
neovim (0.1.9)
pip (8.1.2)
pycrypto (2.6.1)
pyserial (2.6)
pyusb (1.0.0)
PyYAML (3.11)
requests (2.10.0)
setuptools (19.4)
shellescape (3.4.1)
six (1.10.0)
trollius (2.1)
vboxapi (1.0)
wheel (0.26.0)
```

In Python scripts, I am able to perform write operations on an RZRaven USB (set channel, inject packet, etc) without error. However, `pnext()` calls always results in the following error:

```text
Error args: (60, u'Operation timed out')
Traceback (most recent call last):
  File "/usr/local/bin/zbnodecapture", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbnodecapture')
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 726, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1658, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/site-packages/killerbee-2.6.0-py2.7-macosx-10.11-x86_64.egg/EGG-INFO/scripts/zbnodecapture", line 48, in <module>

  File "build/bdist.macosx-10.11-x86_64/egg/killerbee/__init__.py", line 292, in pnext
  File "build/bdist.macosx-10.11-x86_64/egg/killerbee/dev_rzusbstick.py", line 461, in pnext
usb.core.USBError: [Errno 60] Operation timed out
```

Using the same versions of pyusb, pyserial, and pycrypto, the reads do not produce errors if I run Killerbee in Virtualbox on the same Mac.

Pyusb debug output:
```text
2016-06-27 14:26:24,440 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fb7f9489ee0 at 102876210>)
2016-06-27 14:26:24,530 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2016-06-27 14:26:24,531 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2016-06-27 14:26:24,531 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x102876990>)
2016-06-27 14:26:24,532 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,532 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x102876990>)
2016-06-27 14:26:24,532 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x1028bea50>, 1)
2016-06-27 14:26:24,532 DEBUG:usb.backend.libusb1:_LibUSB.reset_device(<usb.backend.libusb1._DeviceHandle object at 0x1028bea50>)
2016-06-27 14:26:24,533 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x1028bea50>)
2016-06-27 14:26:24,533 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x102876990>)
2016-06-27 14:26:24,534 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 128, 6, 768, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2016-06-27 14:26:24,534 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 128, 6, 770, 1033, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2016-06-27 14:26:24,535 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 128, 6, 771, 1033, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2016-06-27 14:26:24,535 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration(<usb.backend.libusb1._DeviceHandle object at 0x102810850>)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 1, 0, 0, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0)
2016-06-27 14:26:24,536 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
[... 5 lines cut ...]
2016-06-27 14:26:24,538 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0, 0)
2016-06-27 14:26:24,538 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0, 0, 0)
2016-06-27 14:26:24,538 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,538 DEBUG:usb.backend.libusb1:_LibUSB.bulk_read(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 132, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 500)
2016-06-27 14:26:24,543 DEBUG:usb.backend.libusb1:_LibUSB.bulk_write(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 2, 0, array('B', [8, 11]), 1000)
2016-06-27 14:26:24,543 DEBUG:usb.backend.libusb1:_LibUSB.bulk_read(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 132, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,  [... cut ...]
2016-06-27 14:26:24,552 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x102876990>, 0)
2016-06-27 14:26:24,552 DEBUG:usb.backend.libusb1:_LibUSB.bulk_read(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 100)
2016-06-27 14:26:24,659 DEBUG:usb.backend.libusb1:_LibUSB.release_interface(<usb.backend.libusb1._DeviceHandle object at 0x102810850>, 0)
```

rmspeers, 2016-07-03:
Is your libUSB backend the same (well, version at least, obviously not exactly the same :)) in the VirtualBox?

sxmichaels, 2016-07-05:
How do I check the exact versions? I see in the log I posted that Mac OS is using libusb 1.0, and Aptitude on Linux tells me that `libusb-1.0.0` and `libusb-0.1.4` are installed, but I can't figure out whow to get a more specific number beyond the major version 1.0 if there is one.

rmspeers, 2016-07-17:
I've googled around to try to find this, and without libusb-config (which appears to not be available on the newer libusbs), I can't tell a good way of finding the version. I think the homebrew approach is good.
If you were working with a pyserial device, like the ApiMote, I'd suggest trying to unload the Apple FTDI drivers that they added, in favor of libusb, but that shouldn't be affecting you here. (https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device)
What confuses me about the above original logs is that you are getting detailed pyUSB logs, but no errors in them. The bulk reads appear to be returning...

brycobat, 2017-01-31:
I'm wondering if the pyusb/libusb API has changed slightly, I've only just started tinkering with an RZ USB stick but the changes at https://github.com/riverloopsec/killerbee/compare/master...brycobat:wip-rzusbstick-mac-fixes are working for me to get sane info back from zbstumbler.

FWIW, I'm on macOS 10.12.3, using brew Python 2.7.13, libusb 1.0.21, libusb-compat 0.1.5.

pip list:
```
appdirs (1.4.0)
crypto (1.4.1)
killerbee (2.6.0)
Naked (0.1.31)
packaging (16.8)
pip (9.0.1)
pycrypto (2.6.1)
pyparsing (2.1.10)
pyserial (3.2.1)
pyusb (1.0.0)
PyYAML (3.12)
requests (2.13.0)
scapy (2.3.3)
setuptools (34.1.0)
shellescape (3.4.1)
six (1.10.0)
wheel (0.30.0a0)
```

(Insert standard disclaimer, almost totally untested, works on my machine, &c.)

rmspeers, 2017-02-26:
Thanks! Is there a reference by chance to the following as an alternative timeout in the libusb docs, or did you just find this empirically?
```
if len(e.args) >= 1 and e.args[0] == 60:  # Alternative "Operation timed out"
```

rmspeers, 2017-02-26:
https://github.com/riverloopsec/killerbee/pull/84

brycobat, 2017-03-13:
Just empirically. (Probably worth trawling the libusb/pyusb docs, but I haven't gotten around to that yet.)

taylorcenters, 2021-06-22:
adding #84  in #228

taylorcenters, 2021-07-01:
This has been long inactive - please use latest version and open a new issue with more recent output.

closed 2021-07-01 by taylorcenters