riverloopsec/killerbee #71: Connection Atmel RZ RAVEN USB Stick and MoteIV Tmote Sky or TelosB

Auron691, 2016-07-21:
Hi, i tried to create a KillerBee device for attack my Zigbee network. I dont want buy all the hardware for create an Raven Usb with KillerBee because it is much expensive, my ideas is to use the Raven Usb with stock firmware to sniff packet and buy 1 MoteIV Tmote Sky or TelosB to create injection. It is possble use this configuration? For flash KillerBee firmware on MoteIV Tmote Sky or TelosB i need more hardware? thanks all

rmspeers, 2016-07-21:
That setup would work. You can flash a Tmote Sky or TelosB using the USB connection.

closed 2016-07-21 by rmspeers

Auron691, 2016-07-21:
thanks for reply, another question if i can. For sniffing  zigbee packet, can i use a usb like this http://www.robot-italy.com/it/xstickr-802-15-4-usb-adapter.html , or i can use only Raven Usb? thanks a lot

rmspeers, 2016-07-22:
We’ve never looked at that device. If you can send it commands, put it into promiscuous mode, etc — then you can probably add support for it in KillerBee by adding a new dev_*.py interface. However, you would have to do that and update the code to support the device.