riverloopsec/killerbee #72: Zigbee sniffing using killerbee channel detecting error

arunmagesh, 2016-07-29:
I configured two Digi xbee s1 module with same channel and PAN ID and both as end device. Communication happens perfectly between these modules.

Problem is when i use rzusbstick with killerbee firmware to look for all the available PAN ID and channel. using zbstumbler it didnt detect any PANID on my channel. when enabled verbose i get "Received frame is not a beacon (FCF=6188). Received frame."

I tried with different channel, PANID and Destination/MY address. Still the same. Is it a problem with the module or the configuration or something else?

adi0x90, 2016-07-29:
Looks like I am having the same issue. Using the Xbee S1 module. 

I added some more debug code and printing out the packet, and below is what I see. 

[image]

joswr1ght, 2016-07-30:
Could it be that the Xbee S1 module is not using standard 802.15.4 Beacon discovery? I don't know anything about that product.

The frames you are showing in the screenshot are data frames. \x61\x88 is a standard 802.15.4 data frame FC header.

-Josh

rmspeers, 2016-08-31:
@arunmagesh or @adi0x90  - Any findings based on Josh's and my input?

adi0x90, 2016-09-11:
@joswr1ght @rmspeers  : Looks like what Josh mentioned "_Could it be that the Xbee S1 module is not using standard 802.15.4 Beacon discovery_" is the actual reason. This might be because S1 is the oldest version of Xbee and **does not directly support Zigbee protocol**, instead just the interception and sniffing of 802.15.4.  

I didn't find a mention of Zigbee in the S1 documentation and datasheet anywhere, so this is what I assume. 

I will soon try it out with the Xbee Pro  and update here.

rmspeers, 2016-11-05:
The zbstumbler command should work with any true 802.15.4 network, even if it doesn't support ZigBee. My understanding is that the S1 may not do standard 802.15.4.

taylorcenters, 2021-06-22:
closed - long idle

closed 2021-06-22 by taylorcenters