riverloopsec/killerbee #74: zbstumbler, zbwireshark and zbdump with problems

Krostas, 2016-10-26:
I'm getting a bunch of errors when running these commands:

**zbstumbler**
Most os the times can get all the equipments but after some time this error appears:

```
Warning: You are using pyUSB 1.x, support is in beta.
zbstumbler: Transmitting and receiving on interface '2:4'
New Network: PANID XXX  Source XXX	
Ext PANID: XXX	Stack Profile: ZigBee Enterprise
	Stack Version: ZigBee 2006/2007
	Channel: XX
Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbstumbler')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 746, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1508, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 191, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 292, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 438, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 334, in sniffer_on
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 306, in _open_stream
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 286, in __usb_write
Exception: Error: Semantical Error
```

**zbdump**

With zbdump I can get some dump file but after a short time I get this:

```
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:4', link-type DLT_IEEE802_15_4, capture size 127 bytes
Traceback (most recent call last):
  File "/usr/local/bin/zbdump", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbdump')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 746, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1508, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbdump", line 78, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 292, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 466, in pnext
IndexError: array index out of range
```

**zbwireshark**

With zbwireshark after a few packets I can't receive anything else.
Some times it freezes at 6 packets, other times 30, 33.
130 packets was the highest I could achieve...

rmspeers, 2016-10-29:
What version of pyUSB and what backend for pyUSB are you running? On what OS/version?

Krostas, 2016-10-30:
I'm using RZUSBSTICK with the second firmware version on the last version of a Kali VM.

About the pyUSB I didn't install nothing... I follow the the steps of the README.

Krostas, 2016-11-04:
I'm using 1.0.0b2 version of pyUSB.
Should I update to the latest version or there is any specific version that I should install?

Thanks

rmspeers, 2016-11-05:
Can you confirm you uninstalled KillerBee that came with Kali and installed the new one from Github? I think this is the case based on your logs but want to confirm.

That pyUSB version should work fine.

Krostas, 2016-11-05:
I didn't uninstall the Kali version, just installed the «new» version.
I will check this and report it later!

Thanks

Krostas, 2016-11-09:
The pyUSB version is 1.0.0b2
I'm using this version of Kali:
`Linux kali 4.6.0-kali1-amd64 #1 SMP Debian 4.6.4-1kali1 (2016-07-21) x86_64 GNU/Linux`

I removed the older killerbee version with:
`apt-get purge killerbee`

Installed everything again and altought everything looks smoother some problems persists.

It seems that **zbstumbler** is better altough sometimes gives the same error.

About the **zbdump** doesn't crashes anymore but freezes during the capture. It capture only 12, 14, 26, 30, (...) packets. No more than 35.

The behavior in **zbwireshark** is the same: freezes after a while.
Can't get no more that 135 packets.
Sometimes gives this error when freezes, other times just stop and nothing happen:
```
root@kali:~# zbwireshark -c 11
Warning: You are using pyUSB 1.x, support is in beta.
zbwireshark: listening on '2:4'
Traceback (most recent call last):
  File "/usr/local/bin/zbwireshark", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbwireshark')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 746, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1508, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 104, in <module>
    
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 81, in main
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 292, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 466, in pnext
IndexError: array index out of range
```

I tried with other computer with the same RZUSBSTICK and the behavior is the same.
I even used other RZUSBSTICK without the killerbee firmware and the wireshark gives the same problem.

I was wondering, since I installed the firmware with the new atmel programmer, do you think that could be a problem of firmware?
After uploading the file of the new firmware I download it again and the code is different from the original. Is supposed to be different?

Should I try with other OS? What do you recommend?

I'm using VMWare! Could this be a problem?

robertfullagar, 2017-03-09:
Krostas

Did you ever get these issues resolved as I am having same/similar frustrations...

Rob

Krostas, 2017-03-09:
RobertFullagar,

Honestly the solution was to get a ApiMote.
It works super fine, can get all the packets and do everything!

Krostas

robertfullagar, 2017-03-09:
Krostas

Where did you get the ApiMote from?

Rob

rmspeers, 2017-03-09:
If you need one, email team@riverloopsecurity.com and we can help (or you can build it off the open-source spec if you want).

Krostas, 2017-03-09:
Robert,

got one from RiverLoopSecurity.
Zero problems!

Scytmo, 2017-06-01:
The issue of zbdump/zbwireshark freezing on both KillerBee and Atmel firmware seems to be a duplicate of #86

taylorcenters, 2021-06-22:
closed - duplicated will continue in #86

closed 2021-06-22 by taylorcenters