riverloopsec/killerbee #75: Sniffing ZigBee packet using RZRAVENUSB stick

Auron691, 2016-11-03:
Hi all, i create a ZigBee network using Xbee series 2 model xb24-z7wit-004.
Devices are program in AT mode, so i try to sniff some packet, zbid command found my RZRAVENUSB, but when i run zbdump, it don't capture any packets. Can you help me? i use only zbid and zbdump command because i know the channel of the network. It must be use zbstumbler command?

rmspeers, 2016-11-03:
Which channel is your XBee network on — and specifically in the 2.4 GHz band? Are you SURE that it is running in 802.15.4 mode?

Auron691, 2016-11-04:
Xbee implementa ZigBee protocollo so they use 802.15.4 , i use Channel 13 and i think that they are use 2.4 GHz of frequency

rmspeers, 2016-11-05:
xb24-z7wit-004 is programmed as a router. Do you have a coordinator in the network that you have it talking to? E.g. is it joined to a network and sending frames? How are you confirming that frames are being sent during the period of sniffing?

Auron691, 2016-11-05:
I have One xbee programmed like a coordinator and One xbee programmed like a router with an Arduino uno to create packets. I am sure because in xctu software i read the packets send by router to the coordinator

rmspeers, 2016-11-05:
Ok -- sounds good. Do you have a raw PCAP from the software that we can look at to see what the frames look like?

Auron691, 2016-11-05:
Yes this is an example of packets:

7E 00 10 17 01 00 13 A2 00 40 3E 17 E6 00 02 02 44 31 05 39

Auron691, 2016-11-07:
Update: today i try to sniff packets and i can sniff only broadcast packets. Where is the error? thanks all

rmspeers, 2018-06-25:
@Auron691  please try sniffing on the 004 version of the firmware and advise if any issues remain.

taylorcenters, 2021-07-01:
long inactive - closing. Please update to latest version and open a new issue if you still have problems. Thank you.

closed 2021-07-01 by taylorcenters