riverloopsec/killerbee #78: pyserial 3 errors

dperret, 2016-11-05:
When using an ApiMote v4 beta device with Kali 2016.2, and Killerbee built from the head revision from git, I receive the following error message when running zbid:

> root@kali:~# zbid
           Dev Product String       Serial Number
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbid')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 746, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1508, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 23, in <module> 

>  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 46, in show_dev
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 224, in devlist
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 303, in isgoodfetccspi
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 134, in serInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 265, in pyserInit
AttributeError: 'Serial' object has no attribute 'setTimeout'
root@kali:~# 

The setTimeout method was removed in pyserial 3.0. See this pyserial issue for details: https://github.com/pyserial/pyserial/issues/66

I tested with an Atmel RZUSBSTICK on the same system, and verified that with the Atmel device, the Killerbee tools work correctly.

Test system: Kali 2016.2 amd64 virtual machine
pyserial version installed: 3.1.1
Killerbee device: ApiMote v4 Beta

Pioquax, 2017-01-11:
I think this issue can be closed, since it was resolved with commit 8740df52bebd354292c99f2b42786394cef3d445.
@dperret: Have you already tested the newest version of Killerbee?

rmspeers, 2017-02-26:
@dperret - Any luck with the new version? If so we can close this out.

dperret, 2017-04-23:
Sorry about the delay. Commit https://github.com/riverloopsec/killerbee/commit/8740df52bebd354292c99f2b42786394cef3d445
 did fix the pySerial 3 errors that I saw with the ApiMote v4.  I tested with pySerial 3.2.1, and again with pySerial 3.3, and killerbee is working with both. Thanks for fixing this.

closed 2017-04-23 by dperret