riverloopsec/killerbee #79: Find network key using ZBDSNIFF

Auron691, 2016-11-22:
Hi all, i create a ZigBee network using Xbee series 2 model xb24-z7wit-004.
I have sniff a lot of packets using AVRUSBSTICK with default firmware and save that in a .dump file.
After that i have converter this into dcf using zbconvert, but when i try to find network key using Zbdsniff, it process 2 files (.dump and .dcf) but it don't find any network key. I am sure that the communication is encrypt.

I attached .dump file if someone would test it.
[Enc.zip](https://github.com/riverloopsec/killerbee/files/606949/Enc.zip)

Thanks for support

Sandb0xED, 2020-05-29:
Hi, I'm trying to use zbdsniff to extract the network key from a pcap (captured from zbdump). However, I keep getting an AttributeError: 'NoneType' object has no attribute 'source'.  Do you have any ideas on how I debug?

[image]

taylorcenters, 2021-06-22:
hi, can you try on the python3 version and update the output?

taylorcenters, 2022-03-08:
Closing as inactive. Please create a new issue if you experience this problem with the current version!

closed 2022-03-08 by taylorcenters