riverloopsec/killerbee #82: Installation on TelosB

Auron691, 2017-01-03:
Hi all, i have a problem, when i try to install killerbee firmware on TelosB mote with the command 
`./flash_telosb.sh`
it return me this error:

`MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "./goodfet.bsl", line 1982, in <module>
    main(0);
  File "./goodfet.bsl", line 1882, in main
    bsl.actionStartBSL()
  File "./goodfet.bsl", line 1164, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "./goodfet.bsl", line 1134, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "./goodfet.bsl", line 800, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "./goodfet.bsl", line 490, in comTxRx
    raise BSLException(self.ERR_RX_NAK)
__main__.BSLException: NAK received (wrong password?)`

Can someone help me? thanks a lot

Pioquax, 2017-01-19:
I replaced the goodfet.bsl with an older version of it from the TinyOS project https://github.com/tinyos/tinyos-main/blob/master/tools/platforms/msp430/pybsl/tos-bsl.in and changed the first line in it to "#!/usr/bin/env python". Then in worked for me (Tmote Sky).

rmspeers, 2017-06-25:
Did this work @Auron691? If so we can look at updating that -- I'm worried about just updating it as the goodfet.bsl does have changes for ApiMote and other H/W.

RosenZhu, 2017-10-19:
@rmspeers 
I ran into the same problem.
I did as @Pioquax said, and it worked. but when I try to use zbdump, it appears
`$ sudo zbdump -c 11 -w hhaw.pcap
See the GoodFET FAQ about missing info flash.
Traceback (most recent call last):
  File "/usr/local/bin/zbdump", line 65, in <module>
    kb = KillerBee(device=args.devstring)
  File "/usr/local/lib/python2.7/dist-packages/killerbee/__init__.py", line 161, in __init__
    raise KBInterfaceError("KillerBee doesn't know how to interact with serial device at '%s'." % self.dev)
killerbee.kbutils.KBInterfaceError: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'.
`
and sometimes, it just froze. What can I do?

baltimoremd, 2017-12-31:
Any resolution?  I know I was able to flash my TelosB's like this in the past (many years ago) but I'm getting this error now.  I'm hesitant to use the older version referenced above...

bigttys0, 2018-08-20:
Having a similar issue. Using API-MOTE v4 BETA. dmesg and lsusb show the device is connected. I didn't try what @Pioquax stated since I wasn't sure if this was applicable to my HW.

**Running ./flash_apimote.sh

Flashing apimotev4_gf.hex to the connected USB serial device.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "./goodfet.bsl", line 1982, in <module>
    main(0);
  File "./goodfet.bsl", line 1882, in main
    bsl.actionStartBSL()
  File "./goodfet.bsl", line 1164, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "./goodfet.bsl", line 1134, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "./goodfet.bsl", line 800, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "./goodfet.bsl", line 479, in comTxRx
    rxHeader, rxNum = self.comRxHeader()        #receive header
  File "./goodfet.bsl", line 385, in comRxHeader
    if not hdr: raise BSLException("Timeout")
__main__.BSLException: Timeout

Any ideas?

fhred, 2019-02-07:
I ran into the same problem, and I don't know what exactly went wrong with the `goodfet.bsl` found in the killerbee repo, but the version from [https://github.com/RIOT-OS/RIOT](url) it worked flawlessly.

I used the following command:

```
board=telosb /path/to/RIOT/dist/tools/goodfet/goodfet.bsl --telosb -e -p /path/to/killerbee/firmware/gf-telosb-001.hex
```

If someone is interested, why that version works, they should look at the diff of both files.

taylorcenters, 2021-07-01:
Thanks for the workaround.

closed 2021-07-01 by taylorcenters

rmspeers, 2021-07-02:
@taylorcenters  did we figure out the diff and/or pull in an updated?