riverloopsec/killerbee #85: Can it be done?

gh0stshell, 2017-03-03:
Could I use the GoodFet to flash my RZ Raven USB Stick (RZUSBSTICK)?

joswr1ght, 2017-03-03:
Not that I know of, but Mike Ossmann did it with a cheap programmer and ingenuity (http://ossmann.blogspot.com/2010/01/killerbee-on-budget.html). I use an AVR Dragon (http://www.mouser.com/ProductDetail/Microchip/ATAVRDRAGON/) and an adapter header (http://www.mouser.com/ProductDetail/Microchip/ATAVR-SOAKIT) and a 10-pin (2x5) 50mm male-male header.

closed 2017-03-03 by joswr1ght

gh0stshell, 2017-03-04:
Would this work ( I have one I use on my drones FC )

http://www.electroschematics.com/9412/usb-asp-programmer-atmega-programming/

I have those items listed above on the way, just trying to see what options I have...

riverloopsec, 2017-03-04:
It looks like it may, but we don’t know. If you get it to work, we’d welcome any notes that we could use to make a tutorial for others!

gh0stshell, 2017-03-06:
> atmega-programming/ <http://www.electroschematics.

Still working on a cheap programming solution

Is the the correct pinout?
http://lth.cow.dk/images/0/04/AVR_Connectors.jpg

gh0stshell, 2017-03-16:
Have some more programmers coming in but already got it down to 90 or less, got the adapter to replace me soldering wires, just tested a new programmer and it works, waiting for the last to arrive to see if they work.

gh0stshell, 2017-03-17:
Currently got it down around $65, still have a few more programmers coming
in and want to give those a try. So far here is the cost list
Adapter - $7
Programmer - $14
RZRAVEN - $ 44

rmspeers, 2017-03-18:
Great news -- thanks for the update.

gh0stshell, 2017-03-24:
Success! First of the under 10$ programmers arrived from a friend that makes them and it works, waiting one 3 more, one of which I am looking forward to
adapter = $7
programmer = $9
raven = $44

rmspeers, 2017-03-24:
Awesome! If that works out I'd love to work with you to have a guest blog post, or a new instruction file added, based on your research!

gh0stshell, 2017-03-25:
There are 3 programmers left that are on their way in this week, 2 I have
high hopes for and once they arrive and are tested I will be creating
something and get parts for about 10 kits, having them on hand would allow
people to get the adapter, programmer and cable in 2-3 days(US). I will
link to the cheapest place to buy the raven, I don't want to have to buy
those.

gh0stshell, 2017-03-30:
Glad I waited, one of the programmers blew up the raven, have a new one coming in today