riverloopsec/killerbee #86: ZBDump, zbwireshark etc problems

robertfullagar, 2017-03-08:
Hi

I have similar issues to another post by Krostas

I bought 2 x RZUSBStick from Farnell and replaced the firmware on both with your 002 firmware using atmel-ice and atmel studio. I have the orange light as described.

I am running Kali Linux, 20016.2 (In VM and USB boot with persistence), Python 2.7.13
, pyusb is version /usr/lib/python2.7/dist-packages/pyusb-1.0.0b2. libusb-1.0-0 is already the newest version (2:1.0.21-1).

I performed a sudo apt-get purge killerbee and then followed your install process.

 zbid shows -
Dev Product String       Serial Number
          2:10 KILLERB001           FFFFFFFFFFFF

when I run.. zbdump -c 20 -i 2:10 -w /tmp/zigbee20.pcap..the capture starts but stops writing to the file after a few kb has been captured but in the run window it appears to be working and shows no errors.  I have many chatty devices on channel 20.  I tried both versions of your firmware with the same result.
Below is me running and stopping the dump when the output file stops growing..notice the small packet count....the ID changes as I was changing firmware etc

zbdump -c 20 -i 2:7 -w /tmp/zigbee20.pcap
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:7', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C103 packets captured
root@kali:~# zbdump -c 20 -i 2:8 -w /tmp/zigbee20.pcap
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:8', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C11 packets captured
root@kali:~# zbdump -c 20 -i 2:8 -w /tmp/zigbee20.pcap
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:8', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C68 packets captured
root@kali:~# zbdump -c 20 -i 2:10 -w /tmp/zigbee20.pcap
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:10', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C124 packets captured
root@kali:~# zbdump -c 20 -i 2:10 -w /tmp/zigbee20.pcap
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on '2:10', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C33 packets captured

Same with wireshark only a few packets are being captured before no more packets arrive in wireshark.
I tried on a stock Kali2016.2 image usingt he installed killerbee tools and got the same results

I am a security consultant and would like to use your tools to test an IoT installation.

Cheers

Rob

robertfullagar, 2017-03-09:
Update

I notice that when these programs are fun and fail the following entry get written to the kern.log

Mar  9 16:12:51 kali kernel: [18817.943096] usb 1-2.1: reset full-speed USB device number 9 using uhci_hcd

Rob

Acesonnall, 2017-04-03:
I can confirm this issue. In addition to the the sniffing process abruptly stopping with either the zbdump or zbwireshark commands, based on what I've gathered by comparing the capture files from the TI CC2531 and the RZUSBStick, the RZUSBstick seems to be missing some frames in the capture file, thus corrupting the format (might notice a "Bad FCS" warning on a number of packets in the Wireshark interface).

(Right side: TI Sniffer | Left side: RZUSBStick)
[image]

Scytmo, 2017-04-06:
@Acesonnall 
> the RZUSBstick seems to be missing some frames in the capture file, thus corrupting
> the format (might notice a "Bad FCS" warning on a number of packets in the Wireshark
> interface).

I have raised this separately as #87 
I don't think it's related to the capture stalling (as the OP reported).

schristiaens, 2017-04-14:
We tracked the issue to packets of length 64 bytes. When a packet of this length was read from the USB device, subsequent reads resulted in zero byte responses. This only happened on 64-byte packets. Packets that were larger were correctly reassembled, and smaller packets had no issue. The problem was difficult to reproduce, as the packet size varied based on the type and manufacturer of the devices in use. 
We attempted to fix this by modifying `pnext()` in `dev_rzusbstick.py` (the read is on line 458) but with no success. Perhaps this is a firmware issue?
We tested on the most recent master branch (last commit showing as April 9th), and were running on Kali using pyUSB 1.0 and firmware 001.
In Wireshark, 64-byte packets show up as "54 bytes on wire", which was slightly confusing. If you look at the screenshot from @Acesonnall, you can see that is exactly the length of the packet where he is seeing the same issue.

Scytmo, 2017-04-14:
Nice find, @schristiaens !
There is special handling in the firmware for multiples of 64 bytes, which I found when I was looking at the truncation issue #87 , but didn't look into in any detail.  You can see it in the firmware for the air capture here:
https://github.com/riverloopsec/killerbee/blob/master/firmware/src/kb-rzusbstick/application/rzusbstick/air_capture.c#L265
There is a comment here "Send Zero Length Packet..." for a section of code if there are no more bytes left to send, but one more packet left to send - which appears to have something to do handling USB transfers that are exact multiples of 64 bytes.  But I didn't get a chance to look into how this interacts with PyUSB and KillerBee.  Needs more investigation to understand whether this is related to the stalling.

Scytmo, 2017-05-22:
I have confirmed, using usbmon, that the RZ USB Stick gets into a state where it continuously responds with zero length packets after it has returned an exact length 64-byte packet.  See a snippet of this below.  I strongly believe that this is related to the zero-length-packet generation that is part of the RZ USB Stick firmware that I referenced above.  The sending of a single zero length frame is normal for USB bulk transfers, as sending a 'short' (less than maximum length') packet is an indication that a USB bulk transfer has ended - but obviously sending these continuously is a bug.  However, I see the same behavior when I re-flash the current version of the Atmel firmware.  I'm still investigating, but I thought I would post this in case it helped.

Snippet of usbmon output (RZ USB Stick on bus 2, endpoint 13).  It is the "C Bi:2:013:1 0 0" lines that are showing a succession of "successful" transfers of zero bytes.  When the RZ USB Stick genuinely does have no data to transmit, it responds with a line containing "C Bi:2:013:1 -2 0" where the '-2' status indicates that the transfer was canceled, as it timed out without any data available):
ffff880035389bc0 2703977436 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2703978619 C Bi:2:013:1 0 15 = 500f7dbf 59020701 050200b5 9e57ff
ffff880035389bc0 2703978851 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2704047370 C Bi:2:013:1 0 64 = 504012d0 59020101 36618805 63f79ee9 00004806 11bc0000 1ed40100 9ee928e7
ffff880035389bc0 2704047714 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2704048621 C Bi:2:013:1 0 0
ffff880035389bc0 2704048783 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2704049817 C Bi:2:013:1 0 0
ffff880035389bc0 2704049952 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2704073814 C Bi:2:013:1 0 0
ffff880035389bc0 2704074151 S Bi:2:013:1 -115 64 <
ffff880035389bc0 2704075503 C Bi:2:013:1 0 0

Scytmo, 2017-05-22:
I have found the bug in the firmware causing this, and fixed it in pull request #92 

I have also (temporarily) put a copy of the built firmware hex file here, so that people can test the fix without having to build their own firmware image:
  https://www.dropbox.com/s/sxg4e1sfuzfdyi6/RZUSBSTICK.hex

As I mentioned above, this bug appears to be in the original Atmel firmware.

Scytmo, 2017-05-30:
@Acesonnall , @schristiaens Can you confirm whether the firmware update I provided above seems to fix the problem you were seeing with zbdump/zbwireshark stopping capturing packets after a while (i.e. as soon as an exact length '54 byte' packet is received).

schristiaens, 2017-05-31:
Awesome @Scytmo! A colleague is borrowing my kit right now, so I am unable to test, but will do so as soon as I get it back. Thanks for taking the time to look at it!

rmspeers, 2017-05-31:
I've also asked someone on our team to validate -- thanks!

Scytmo, 2017-06-16:
Any news, @rmspeers @schristiaens ?  FWIW, I've been using the patched firmware almost daily for the last 2-3 weeks without issues (and without seeing it stall, as it did previously).

rmspeers, 2017-07-10:
@Scytmo - My fault, I thought this was merged but now see it's another PR than the one that was linked here. In https://github.com/riverloopsec/killerbee/pull/92, Ricky from the RLS team has said it looks good. I will merge it. Please let me know if this is then good to close.

mrnice1, 2018-08-29:
@Scytmo : I have the same issue as above. I already have the latest firmware 006 flashed on the rzusbstick device, but I get the same errors with zbstumbler, etc.

Any ideas?

rmspeers, 2018-08-29:
Version 006 should include this patch for the 64 byte issue, but @Scytmo and @AdamLaurie may know off-hand, else I can check the code.

Scytmo, 2018-08-29:
@rmspeers I just checked the 'master' branch, and the fix is included.
@mrnice1 You mean with zbdump/zbwireshark stopping capturing packets after a while (i.e. as soon as an exact length '54 byte' packet is received)?  No ideas, I'm afraid, as that (very specific) problem was completely explained/fixed, as far as I'm aware.  Are you sure you are seeing exactly the same thing?  (i.e. the capture stalling after a while, with the last packet being 54 bytes in wireshark).

AdamLaurie, 2018-08-29:
@mrnice1  Please provide your exact command line.

mrnice1, 2018-08-29:
I don't know what I did right now and I'm getting a wholly new error:
sudo zbid

Dev Product String       Serial Number
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbid')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 719, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1511, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 23, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 46, in show_dev
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 290, in devlist
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 355, in isgoodfetccspi
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 134, in serInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 187, in pyserInit
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 180, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 311, in open
    self._update_dtr_state()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 605, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 22] Invalid argument

mrnice1, 2018-08-29:
BTW: I'm installing Ubuntu 16.04.5 (32bit) on a virtual machine right now. Maybe it'll work then.

rmspeers, 2018-08-29:
The second error you have seems to be as it is trying to connect to a serial device and perhaps pyserial versions are miss-matching. Is it possible that between your first and second run you plugged in some serial-enabled device to USB or the computer?

mrnice1, 2018-08-29:
@rmspeers I didn't plug in anything. I tried updating pyUSB with pip and so on. Maybe the problem accurs because of that?

mrnice1, 2018-08-29:
Now I tried the following commands for reinstalling python:

sudo apt-get remove python
sudo apt-get install python
sudo apt-get install python-gtk2 python-cairo python-usb python-crypto python-serial python-dev libgcrypt-dev

After typing sudo zbid, I get this error:
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbid')
ImportError: No module named pkg_resources

mrnice1, 2018-08-29:
@rmspeers @AdamLaurie @Scytmo : Now I got it finally working again as it did. 
But I still get this error message:
sudo zbstumbler

sudo zbstumbler
Warning: You are using pyUSB 1.x, support is in beta.
zbstumbler: Transmitting and receiving on interface '1:7'
Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbstumbler')
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 661, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 191, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 357, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 546, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 385, in sniffer_on
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 355, in _open_stream
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 334, in __usb_write
Exception: Error: Semantical Error

AdamLaurie, 2018-08-29:
I am seeing the same problem with zbstumbler - I'll look into it.
However, can you run e.g.
zbwireshark -c 11
or
zbdump -w /tmp/x.pcap

mrnice1, 2018-08-29:
@AdamLaurie nothing worked. But I think it's because of my system. I installed Ubuntu on another Computer and everything worked fine, instead of #143

taylorcenters, 2021-07-01:
long inactive - closing. Please update to latest version and open a new issue if you still have problems. Thank you.

closed 2021-07-01 by taylorcenters