riverloopsec/killerbee #91: zbstumbler error : ERROR: Failed to set channel to 11. (tuple index out of range)

txjoe, 2017-04-26:
I encountered this error while trying to run zbstumbler on an ubuntu 14.04 vm. This seems to work fine on my 16.10 vm. Any ideas ?

Error

```

zbstumbler: Transmitting and receiving on interface '002:007'
ERROR: Failed to set channel to 11. (tuple index out of range)
```

sgript, 2017-06-15:
I'm having this exact same issue, were you able to find any solutions?

Scytmo, 2017-06-16:
This seems to be due to the version of python-usb/pyusb included in the Ubuntu 14.04 distribution (version 0.4.3).  You can get around this by removing version 0.4.3 installed by apt, and installing the lastest version 1.0.0 using pip:

Remove the version 0.4.3 installed by apt: "sudo apt remove python-usb"
Install the latest version using pip: "pip install pyusb --user" (I prefer to use the "--user" option to install it in under my $HOME directory. YMMV)
You may have to install pip if you don't already have it: "sudo apt install python-pip"

This should fix the problem.

sgript, 2017-06-20:
This solved it, thanks @Scytmo!

closed 2018-11-07 by rmspeers