riverloopsec/killerbee #93: TelosB compatibility question

sgript, 2017-06-18:
Hi there,

I was going to use an RZ Raven USB for sniffing and then buy the MTM-CM5000-SMA "based" on TelosB, would this work for packet injections using killer bee / flashing firmware?

I was looking for the TelosB mote and this is all I came across, which I need for a project of mine.
https://www.advanticsys.com/shop/mtmcm5000msp-p-14.html

rmspeers, 2017-06-19:
Replied via email as well, but to track it here also: We haven't used this hardware so we can't say for sure. It looks like it would probably work, but please report here if you try it so we can update the list of hardware for others.

closed 2017-06-19 by rmspeers

rmspeers, 2017-07-02:
The vendor linked above has Tweeted to us to assure us it works. I will update the README file to link to it.