riverloopsec/killerbee #97: Issues with zbfind

sgript, 2017-07-03:
I've come across this issue where when using zbfind nothing actually happens. As I have seen this mentioned somewhere else also, I was wondering what the issue was (if it's known) and if it's possible for me to get it to work?

[image]

rmspeers, 2017-07-04:
Please try out zbid and zbwireshark, and then zbstumbler. Run them all as sudo given that is how you are running zbfind. It looks like your computer isn't able to connect to a KillerBee device correctly.

sgript, 2017-07-04:
Is there any way I can fix that? I am currently using Ubuntu 16.04 natively installed on a HP Pavilion dv6. Issue is with

rmspeers, 2017-07-05:
@sgript have you tried out zbid, zbwireshark, and zbstumbler? What are those doing for you? I want to see if those are detecting and properly connecting to your device.

sgript, 2017-07-13:
@rmspeers I had issues getting hold of the TelosB Mote from Advanticsys, so I'm using an RZ Raven USB, flashed. All three of the tools you mentioned are working.

closed 2018-03-18 by sgript