riverloopsec/killerbee #99: When can i expect injection and jamming in TI cc2531 stick?

ankitamarwah, 2017-07-12:
I got the TI CC2531 Dongle and i could only sniff packets. I really can't replay anything back to the channel like apimote or rzusbstick. 
just want to know when I can expect that update. 
I really can't do anything without Killerbee.

rmspeers, 2017-07-13:
The TI CC2531 dongle support is sniffing only at this time.

The README lists what dongles support what capabilities. RZUSBSTICK with KillerBee firmware supports injection, as does the ApiMote or TelosB currently.

CC2531 injection support is not being worked on at this time.

nteklik, 2018-10-30:
Hello,
 I have TI CC2531 dongle and when I run zbid I got "The device has no langid". I tried to run zbid on Vbox linux ubuntu(win10host) and Linux host machine(Ubuntu). Do I need to flash cc2531 with another hex file?or, What should I do.
Regards,
*If you need I can attach the screenshot of the error.

nteklik, 2018-11-15:
> I got the TI CC2531 Dongle and i could only sniff packets. I really can't replay anything back to the channel like apimote or rzusbstick.
> or any other alternative you can recommend? like raspbee or zigbit?
> just want to know when I can expect that update.
> I really can't do anything without Killerbee.
Hello,
I am not at the same situation. How did you solve your problem? 
Regards,
Nedim

rmspeers, 2018-11-15:
There is no one actively working on injection firmware for the TI CC2531, but we expect it shouldn't be tough to be added.

Per my above note, the README lists what capabilities are available in other devices.

nteklik, 2018-11-15:
> There is no one actively working on injection firmware for the TI CC2531, but we expect it shouldn't be tough to be added.
> 
> Per my above note, the README lists what capabilities are available in other devices.

I have read README. Some of them they are so expensive on the other hand, it is almost impossible to buy some of them anymore. They are sold out. But can we transmit and receive data with cc2531?

nteklik, 2018-11-15:
> Per the above answers, you can't transmit currently on the CC2531 with the firmware available. As I have explained, there is no one that I know of currently working on firmware to add injection to this platform, but I expect it should be easy to add. I suggest looking at the datasheet and firmware source code and look at contributing injection features, if you want it supported soon (or I guess you could contract someone to develop it for you if you want it supported ASAP and don't want to add it, feel free to email if so).

First of all thaknks for your responses and help.Which part of killerbee I need to examine&understand for injection with cc2531. I will also get help from my friends. But maybe there is a section in the source code of killerbee for sniffing(receiving the frames) and another section the transmitting the frames. I will examine those sections. And If,most probably and I guess as you mentioned I will need to examine datasheet of cc2351. Could you please guide me little bit more with basic steps in general(summary).Regards...

rmspeers, 2019-02-04:
I suggest anyone looking to modify the CC253x firmware to add injection support work off of any available TI example code and build from there.

closed 2021-06-22 by taylorcenters