riverloopsec/killerbee #102: run zbfind show "raise e"

ooqweasdzxc, 2017-08-05:
When I tried to run zbfind, I got "raise e".I'm using a Raspberry Pi 3B.

pi@KZ:~ $ sudo zbid
           Dev Product String       Serial Number
       001:016 RZUSBSTICK           2EAAF6A01C25
pi@KZ:~ $ sudo zbfind
Error args: ('\xe8\xbf\x9e\xe6\x8e\xa5\xe8\xb6\x85\xe6\x97\xb6',)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-armv7l.egg/EGG-INFO/scripts/zbfind", line 933, in run
  File "build/bdist.linux-armv7l/egg/killerbee/__init__.py", line 298, in pnext
    return self.driver.pnext(timeout)
  File "build/bdist.linux-armv7l/egg/killerbee/dev_rzusbstick.py", line 454, in pnext
    raise e
USBError: Connecting time out

What should I do?
Thanks!!

riverloopsec, 2017-08-05:
Can you try using zbdump and see if that gives you issues?

It looks like the USB stack dropped the device. I’m not sure people have tested on recent RaspberryPi versions, so you may also want to look at dmesg to troubleshoot pyusb if you can isolate the issue to that.

ooqweasdzxc, 2017-08-08:
riverloopsec, thank you.
I'm busy these days, and I forgot my Pi at home.

I used my new AVRDRAGON to update the firmware killerbee-002 and I tried to use VMmare 12.5.2 with Ubuntu14.04.5 i386, but when I tried to use zbfind, I got the error as same as my RaspberryPi. 

dmesg in Ubuntu14.04.5 showed:
[   83.199966] usb 1-2.1: Product: KILLERB001
[   83.199968] usb 1-2.1: Manufacturer: ATMEL
[   83.199969] usb 1-2.1: SerialNumber: FFFFFFFFFFFF
[   96.498554] usb 1-2.1: reset full-speed USB device number 4 using uhci_hcd

and pip list showed:
chardet (2.0.1), crypto (1.4.1), killerbee (2.6.0), pycrypto (2.6.1)
python-debian (0.1.21-nmu2ubuntu2)
pyusb (0.4.3)
PyYAML (3.12)
scapy (2.2.0-dev)
shellescape (3.4.1)

And when I use zbdump, it runs with no response, when I Ctrl+C, shows usb.USBError: Connection timed out.
~$ sudo zbdump -f 15 -w savefile.dump
zbdump: listening on '001:004', link-type DLT_IEEE802_15_4, capture size 127 bytes
^C-110
Traceback (most recent call last):
  File "/usr/local/bin/zbdump", line 5, in <module>
    pkg_resources.run_script('killerbee==2.6.0', 'zbdump')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1401, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-i686.egg/EGG-INFO/scripts/zbdump", line 78, in <module>
    
  File "build/bdist.linux-i686/egg/killerbee/__init__.py", line 298, in pnext
  File "build/bdist.linux-i686/egg/killerbee/dev_rzusbstick.py", line 449, in pnext
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-i686.egg/EGG-INFO/scripts/zbdump", line 20, in interrupt
    
  File "build/bdist.linux-i686/egg/killerbee/__init__.py", line 253, in sniffer_off
  File "build/bdist.linux-i686/egg/killerbee/dev_rzusbstick.py", line 344, in sniffer_off
  File "build/bdist.linux-i686/egg/killerbee/dev_rzusbstick.py", line 313, in _close_stream
  File "build/bdist.linux-i686/egg/killerbee/dev_rzusbstick.py", line 263, in __usb_write
usb.USBError: Connection timed out

Why my SerialNumber is "FFFFFFFFFFFF"? And which step I had make a mistake?

rmspeers, 2018-06-02:
The firmware version just sets serial number to that -- no one has made an update to change that to something unique.

I would suggest using the new RZUSBSTICK firmware now, 003, and see if that helps. Perhaps this is related to the packet-length-specific issue which that firmware observed.

closed 2021-06-22 by taylorcenters