riverloopsec/killerbee #104: command not find

RosenZhu, 2017-09-11:
I follow the instruction to install killerbee. but when I've installed it, I tried some commands and it showed "command not find".
`zbfind: command not found`
Linux system is Ubuntu 16.04 and it's installed on a virtual machine (parallels desktop) . 
Python 2.7

rmspeers, 2017-09-12:
This seems like and installation issue. Are any commands installed? What is your PATH variable set to? Do any directories in your path include the scripts? Please also rerun the install and post the full command line and output for that, as well as your PATH.

RosenZhu, 2017-09-13:
I install the following version of killerbee and the commands can be found.
[https://github.com/attify/Attify-Zigbee-Framework](url)

rmspeers, 2017-09-14:
Ok, all that installer does is to download killerbee and then run our install script, so I think that likely when you tried the KillerBee install script the first time something went wrong -- perhaps permissions. The console log should have indicated this. I'll close this though given it seems like you're set.

closed 2017-09-14 by rmspeers