riverloopsec/killerbee #107: zbwireshark: root privileges cause Wireshark error

jrussell88, 2017-10-27:
I'm running zbwireshark with root privileges, otherwise it fails with various errors, however when Wireshark opens it displays the error:

> Lua: Error during loading:
 [string "/usr/share/wireshark/init.lua"]:44: dofile has been disabled due to running Wireshark as superuser. See https://wiki.wireshark.org/CaptureSetup/CapturePrivileges for help in running Wireshark as an unprivileged user.

So Wireshark should be run with user privileges. 

Is there any way round this?

rmspeers, 2017-10-28:
The right route is probably to let zbwireshark run as the user, and thus Wireshark will also be run as the user. I assume the errors you are getting when running zbwireshark as the user relate to not having permissions for your user to access the USB or serial devices, is this the case? Which error are you getting, and which device is in use?

jrussell88, 2017-10-29:
Here's what I get running as user:
> $ zbwireshark -c 23
Warning: You are using pyUSB 1.x, support is in beta.
Traceback (most recent call last):
  File "/usr/local/bin/zbwireshark", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbwireshark')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 104, in <module>   
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 59, in main    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 111, in __init__
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 127, in __init__
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 160, in __handle_open
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 169, in __dev_setup_v1x
Exception: Unable to open device. Ensure the device is free and plugged-in. You may need sudo.

It looks like it can't access the USB device.

rmspeers, 2017-10-29:
Ok, instead of sudoing (yes, I know the message says that), instead try to troubleshoot the permissions on the device. One example of how to do this in one case is at https://github.com/GoldenCheetah/GoldenCheetah/wiki/Allowing-your-linux-userid-permission-to-use-your-usb-device, but this depends on OS, and there are multiple ways to do it.

taylorcenters, 2021-06-22:
closed - long idle

closed 2021-06-22 by taylorcenters