riverloopsec/killerbee #111: zbwireshark not capturing anything

seed1234, 2017-11-29:
Hello,
I have a RZUSBSTICK i am using the stock firmware that came with it, I didnt know i required a ATMEL ATAVRDRAGON to flash killerbee firmware on it.

Now the issue i am facing is the zbwireshark command is not showing any output on any channel the wireshark gui is blank.

zbid shows the RZUSBSTICK 

https://imgur.com/a/faRY8

rmspeers, 2017-11-30:
Please confirm the channel is correct to what it is sending on; confirm the channel is in the right GHz space -- e.g. US vs European channels! -- and then try zbdump to see what it gets

seed1234, 2017-12-01:
Thank you for your reply,I figured out the issue was the front usb was not functioning properly and would not connect to the RZ USBstick.The zbwireshark works great and i can see the packets.The RZ USBstick is using its stock firmware. I have ordered RZDragon to flash it to killerbee.

I am using to Xbee S1 devices connected to each other one is the coordiantor and the other is the end device.In the XCTU configuration page i have setup the devices to use AES Encrytion with the 32 Hexadecimal key.When i connect the 2 devices i can see the communication packets in the zbwireshark, however I do not see any transport key being transferred between them .I have set up the devices to send "Hello World!" in a infiniate loop and i can see the traffic via wireshark.
Do these devices transport the AES key in plain text or is it saved in the devices?.

joswr1ght, 2017-12-01:
Key provisioning can be done in several ways in ZigBee. Often, the keys are hard-coded in the firmware of devices so the key is not shared over the network. Other times the key is delivered over the network in plaintext, or derived through a process called SKKE.

You'll have to investigate your particular implementation to figure out how the key is established. If it is embedded in the firmware for both devices, you might be able to recover it using `zbgoodfind`.

seed1234, 2017-12-01:
Thanks for the help

closed 2021-06-22 by taylorcenters