riverloopsec/killerbee #114: zbkey error

b1rdi3, 2018-01-22:
Receive the following error when trying to use zbkey:

zbkey -f 15 -p 0xC4D8 -a 9748ccd83d78c550 -i 3:5Traceback (most recent call last):
  File "/usr/local/bin/zbkey", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.6.0', 'zbkey')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 742, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1510, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.6.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbkey", line 131, in <module>
    
TypeError: Non-hexadecimal digit found

I ran all install commands under sudo. Thoughts?

KaloyanDimitrov, 2018-03-16:
Skip the 0x in front of the pan ID

rmspeers, 2018-03-29:
Thanks @KaloyanDimitrov  -- @b1rdi3  did that work for you now?

b1rdi3, 2018-03-30:
@KaloyanDimitrov thanks. 

@rmspeers it works, so Im closing out the issue.

closed 2018-03-30 by b1rdi3