riverloopsec/killerbee #116: AVR Wireless Software (Windows) Flashing method incompatible with 64bit Windows

endiz, 2018-02-09:
I was having a really hard time flashing killerbee on the RZUSBSTICK. I was trying to follow @jrussell88 method posted on the firmware readme.md, but couldn't get the firmware upgrade to start. Each time I started the upgrade process from AVR Wireless Services, the upgrade would crash and say "rfservicesserver.exe has stopped working" and AVR Wireless Services would error out saying the device could not be connected.

Turns out the 64bit drivers don't install properly. You need a **32bit OS with the 32bit drivers** in order to successfully flash the killerbee firmware. Can someone please update the firmware readme.md to reflect this so other people don't get stuck?

[image]

rmspeers, 2018-02-09:
Thanks for the input! Feel free to submit a PR or I will update these documents when I do another update.

endiz, 2018-02-09:
Done https://github.com/riverloopsec/killerbee/pull/117

fecaleagle, 2018-02-15:
I was able to flash the `kb-rzusbstick-002.hex` firmware using this method, but not the `kb-rzusbstick-003.hex` firmware on 64-bit Windows.  I needed to extract the 64-bit wdreg executable from the latest Jungo WinDriver installer (evaluation version), after which I was able to install the 64-bit driver windrvr6 using the following:

`wdreg.exe -inf "C:\Program Files (x86)\Atmel\AVR Tools\usb64\windrvr6.inf" install`

Bummed I can't flash the -003 firmware without an AVR Dragon or fashioning the connectors for the Bus Pirate, but I wanted to let you know that if you can get the latest 64-bit wdreg.exe from a WinDriver evaluation copy, you can install the drivers and flash the -002 firmware.  Now that -002 is flashed, I can't try again with -003 until I revert it back to stock with the Bus Pirate, and I wish I had grabbed the error message, but it was complaining about an invalid firmware format with -003.

closed 2021-06-22 by taylorcenters