riverloopsec/killerbee #119: SEWIO Open Sniffer with Killerbee

antoineQue, 2018-04-12:
Hello everyone !

I encounter some issues using Killerbee with my Open Sniffer from SEWIO.

Sniffer:
-Firmware version 0.9.0
-IP:10.10.10.2

Windows 10, host:
-IP: 10.10.10.11

Kali VM:
4.15.0-kali2-amd64
-IP: 10.10.10.12

The Sniffer is connected with ethernet on my Windows. My kali VM is configured in bridged mode, with VMware workstation. All the nodes can ping themselves and I have access to the web configuration page of the sniffer via my Kali.

My issue is: Killerbee doesn't want to recognize my sniffer. How can I use it with this sniffer ?
I uninstalled KillerBee that came with Kali and installed the new one from Github. 

[image]

[image]

[image]

Thanks for the help,

AQ

Crypt0s, 2018-05-04:
@antoineQue What hardware version of the SEWIO do you have? I noticed they changed hardware some time in the recent past.

antoineQue, 2018-05-04:
I have the OpenSniffer v3.0 or gen3, the latest version available in their shop.

rmspeers, 2018-05-04:
I think I have one of those, as my original version got lost in the mail when I loaned it to someone, so I should be able to recreate this. I assume it's something like the version number coming back is different than what the tool expects so it raises an error.

antoineQue, 2018-05-14:
Hi @rmspeers , do you have some news about the implementation ? Do you want some help for the development or more information that I can give you with my sniffer ?

rmspeers, 2018-05-14:
It looks like they now publish an alpha python package that can be used to communicate instead of having to do it directly to the UI/etc as we originally had coded it. I think using this is likely "better" than doing it directly via potentially unsupported APIs.

I am currently updating our docker build (alpha) to install this library, which isn't on package managers at all.

The next step would be to convert all the logic in the dev_sewio.py file over to call down to this library, and then test that.

rmspeers, 2018-05-15:
Unfortunately the library that Sewio provides is Windows only, and needs to be improved to be able to work cross-platform. For example, calls to exec the Windows netsh command need to be switched to use code like from https://stackoverflow.com/questions/166506/finding-local-ip-addresses-using-pythons-stdlib to be cross platform.

Also, things initialized at import (local listening on address) probably need to be moved into the class so IPs can be configured as needed when auto-discovery doesn't work.

rmspeers, 2018-05-15:
@antoineQue - I've pushed my work in progress to the branch `feature/119-sewio-v09`. This sniffer requires that we hit HTTP endpoints, then parse the results to grab info out of JS/HTML tags, and then receive packets via a UDP interface we bind on. Thus, the API is pretty hacky so that branch still warns the firmware version is unsupported. Can you test that please and see if anything else is breaking??

rmspeers, 2018-05-15:
I'm unable to test in Docker, likely as I have something wrong with the IP networking. I'm recording what I did here for reference.
```
$ docker network create -d macvlan --subnet=10.10.10.0/24 --gateway=10.10.10.1 -o parent=eth0 pub_net
$ docker run -v "`pwd`":/opt/killerbee -v"`pwd`/..":/root -i -t  --network pub_net --ip 10.10.10.3 killerbee
```
I believe the issue is this doesn't give the container a route via 10.10.10.1 on your host.

rmspeers, 2018-05-15:
@antoineQue  I was able to verify on an OSX host connected to my FW v0.9 Sewio that sniffing is functioning. I have marked that as a tested version.

KillerBee doesn't yet implement injection, which the newer version appears to support, as well as jamming. I suggest we tackle new features in a separate ticket.

Please verify this branch now works for you for sniffing and channel tuning, and then we can merge those updates and close this.

antoineQue, 2018-05-15:
Well, first, thanks for all the information and improvements.
Here a screen of what I can do (or not):

[image]

Zbid detect now the sniffer and my firmware version !
I guess zbstumbler can't work because injection is not enabled yet.
About, zbdump, no trames are captured, that's weird because wireshark can (also on channel 15).

I don't really understand why the address 10.10.10.1 is targeted and not reachable (but with ping it's ok).

null0x00, 2018-05-16:
Did a PR.
The injection method was tested with `zbreplay` and `zbstumbler`.
Jamming methods are yet to be properly tested.

Please verify that this changes work properly as this was a rather rushed implementation.

rmspeers, 2018-06-10:
@null0x00  I have merged yours and updated a bit in my branch.

@null0x00  or @antoineQue - I do want to hear if you're getting clean PCAP with zbdump and that you're getting networks found when using zbstumbler using the most up-to-date code from the branch `feature/119-sewio-v09`.

antoineQue, 2018-06-12:
Well, the branch feature/119-sewio-v09 doesn't seem to work for me. Nop packets is captured on the right channel. Actually, I'm working with @null0x00 fork. But I made some changes for the replay, that I will indicate in your branch.

whoot, 2018-11-01:
Hey,
in Europe ZigBee is using 868 MHz as a frequency - and I think it is using just one channel. Is this supported?
*zbstumbler* is just cycling through channels 11 to 26, which are used in 2.4 GHz. Am I missing something?

rmspeers, 2018-11-01:
Which capture device are you using? You'll need to update zbstumbler to tell it to use the page and channels to handle that frequency.

rmspeers, 2018-11-01:
@antoineQue  just wanted to confirm from your June post -- is master of this KillerBee repo working for you now? Were there other changes that were needed to make it work that we need to update?

whoot, 2018-11-01:
I´m using the SEWIO OpenSniffer in v3 (firmwareversion 0.9.0), which is why i wrote the question in this thread

rmspeers, 2018-11-01:
Ok, the above answer should cover it -- updating zbstumbler for the page/channel you want. If you need assistance or pointers, please open an issue on this specific topic and we can assist further.

taylorcenters, 2021-07-01:
long inactive - closing. Please update to latest version and open a new issue if you still have problems. Thank you.

closed 2021-07-01 by taylorcenters