riverloopsec/killerbee #120: GPS not working in ZBWarDrive

bushido8989, 2018-04-29:
Howdy!

Platform is a Nexus 5 running Kali Nethunter, GPSD running and verified (using cgps and testgps.py) with satellite fix.

Everything else in the framework appears to work fine. Can dump packets; zbstumbler works fine. 

Installed via cloning the repo. Error returned when using the "-g -v" flags:

------

WARNING: No route found for IPv6 destination :: (no default route?)
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "build/bdist.linux-armv7l/egg/killerbee/zbwardrive/zbwardrive.py", line 21, in gpsdPoller
    gpsd = gps.gps()
AttributeError: 'module' object has no attribute 'gps'
Warning: You are using pyUSB 1.x, support is in beta.

---

Any assistance would be great!!

rmspeers, 2018-06-02:
This appears as the gps module doesn't have a function called gps. That is odd as it looks like that function exists. If you used `https://github.com/riverloopsec/killerbee/blob/master/killerbee/zbwardrive/testGPS.py` to test, then that is doing the same call, however as you will see there. Please ensure you are testing with that file and also note the directory you are running each command from -- my guess is it's something like that.

taylorcenters, 2022-03-08:
Closing inactive. Please check out the latest version that has updated to Python3. Open a new issue if you still run into problems! thanks!

closed 2022-03-08 by taylorcenters