riverloopsec/killerbee #128: please tag 2.0 release

blshkv, 2018-06-11:
It was announced on the website, however it is not available for download.
Please release a proper tar.gz file.

rmspeers, 2018-06-11:
Unless there is a specific need for the 2.0 in a tar.gz, perhaps it's best if we just do this for the upcoming 2.7 once it is merged down.
May I ask what the tar.gz is needed for as typically the install for most users is a git clone and then a setup.py based install. Is it for packaging into a distro or similar?

blshkv, 2018-06-11:
You are right, it is useful for all distro to pickup the latest version. I'm Pentoo developer and pushed the latest snaphost here:
https://github.com/pentoo/pentoo-overlay/tree/master/net-wireless/killerbee

However, other distros are still on the version 1.0, see:
https://repology.org/metapackage/killerbee/versions

closed 2021-06-22 by taylorcenters

blshkv, 2021-06-24:
just a little "thanks" would be nice before closing the bug

taylorcenters, 2021-06-24:
Sorry - I went on autopilot while trying to clean up the project. Thanks for your help on this one!