riverloopsec/killerbee #130: unable to setup

6d6176, 2018-06-21:
python setup.py
No setuptools found, attempting to use distutils instead.
WARNING: No route found for IPv6 destination :: (no default route?)
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'install_requires'
  warnings.warn(msg)
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied

rmspeers, 2018-06-21:
This appears to just be a standard python issue -- can you try doing an install such as "python setup.py install" as typical?

6d6176, 2018-06-21:
python setup.py install
No setuptools found, attempting to use distutils instead.
WARNING: No route found for IPv6 destination :: (no default route?)
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'install_requires'
  warnings.warn(msg)
running install
running build
running build_py
running build_ext
building 'zigbee_crypt' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-nbjU53/python2.7-2.7.15~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/include -I/usr/include -I/sw/include/ -Izigbee_crypt -I/usr/include/python2.7 -c zigbee_crypt/zigbee_crypt.c -o build/temp.linux-x86_64-2.7/zigbee_crypt/zigbee_crypt.o
unable to execute 'x86_64-linux-gnu-gcc': No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

6d6176, 2018-06-21:
i had to install these 

sudo apt-get install -y libxml2-dev libxslt1-dev zlib1g-dev python3-pip
sudo pip3 install lxml

as well as run sudo python setup.py install

6d6176, 2018-06-21:
also, is there a doc on the usage? I'm trying to get it to work with the atmel RZUSB stick but it doen't seem to find it, tho I have it showing up in the VM now

6d6176, 2018-06-21:
kb@ubuntu:~/killerbee$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 006: ID 0451:16ae Texas Instruments, Inc. 
Bus 002 Device 005: ID 03eb:210a Atmel Corp. 
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

rmspeers, 2018-06-21:
What is the output from zbid? Have you reviewed the tools and README file?

6d6176, 2018-06-21:
I appreciate your help. I have gotten it all to work except for packet injection now. Also have a weird wireshark permissions problem where it says I can’t use as sudo. But thanks for all the help!

rmspeers, 2018-06-21:
What does zbid show? If it doesn't show the KB firmware, injection won't work.

6d6176, 2018-06-21:
It shows that I have the ti 2350 stick and the atmel usb it doesn’t say anything about killer be. I’m not around my machine until Monday tho. It does not show it as KB tho. Would that mean I need to reflash the atmel?

rmspeers, 2018-06-25:
Yes, you do not have one flashed with the correct firmware. There is a new firmware in a branch, 004, that should merge shortly which has some new features.

rmspeers, 2018-06-25:
See PR https://github.com/riverloopsec/killerbee/pull/131. You can get it in the branch now, or it should be merged soon into master.

6d6176, 2018-06-25:
in what branch? I'm not able to clone it from git.

6d6176, 2018-06-25:
i was able to get firmare-led but its not populated

rmspeers, 2018-06-26:
It is now merged to KillerBee master, so try now.

6d6176, 2018-06-26:
Thanks man you’ve helped a lot. I’m not waiting for a bus pirate in the mail so I shouldn’t be bugging you for a few days lol.

closed 2018-11-07 by rmspeers