riverloopsec/killerbee #135: Flashing/openocd + buspirate

6d6176, 2018-07-09:
killerbee@ubuntu:~/Desktop/openocd-0.9.0$ openocd -f openocd_bp.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
adapter_nsrst_delay: 100
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: Invalid command argument
embedded:startup.tcl:21: Error: -chain-position required when creating target
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "openocd_bp.cfg", line 42
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21

6d6176, 2018-07-09:
I have no clue what is going on here, I was following the guide in the READme and the links are broken so i found the openocd software and loaded it up so i think but I am unable to trace the bugs down in the doxygen.

taylorcenters, 2021-06-22:
Closed - please reopen with more details about what you are trying to do if this is still a problem.

closed 2021-06-22 by taylorcenters