riverloopsec/killerbee #137: ZBReplay :UnboundLocalError: local variable 'response' referenced before assignment

muralikrishnaj, 2018-07-17:


muralikrishnaj, 2018-07-17:
While i'm running zbreplay it throwing error like below:

DEBUG: Received operation timed out error ...attempting to continue.
Traceback (most recent call last):
  File "./zbreplay", line 86, in <module>
    kb.inject(packet[0:-2])
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 288, in inject
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 424, in inject
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 284, in __usb_write
UnboundLocalError: local variable 'response' referenced before assignment

rmspeers, 2018-07-17:
Which version of firmware are you running on the rzusbstick? What version of Killerbee are you running?

muralikrishnaj, 2018-07-17:
@rmspeers Thanks for the quick response. Sort it out by changing the value of injecting packet like
kb.inject(packet[0:-2]) to kb.inject(packet[0:-1])

closed 2018-07-17 by muralikrishnaj