riverloopsec/killerbee #139: Wireshark Data

NickEONeill, 2018-07-21:
When testing the zigbee-network-key sample data with zbdsniff I can see the Network key, both before and after converting this data to a pcap file with zbconvert, however when I'm trying to check the data in Wireshark, all of the data shows as 'Bad FCS' is this an issue with the currently version of Wireshark reading the data, or is there something I'm missing? I have tried changing the security level, and also tried it with including the network key provided from zbdsniff. The reason I was testing the sample data was that I was having similiar issues with the wireshark data that was being captured with the RZ Raven stick.

[image]

AdamLaurie, 2018-07-24:
The source packets all have their FCS set to FFFF (this is a print of the packet from the zbconvert util as it's being read in):
[[1193806329.95552, 60, 60], 'a\x88\xdf.\x1b>\x14\x9c\x17\x08\x00\x01\x00+\x18\x08\x0e\x00d\x06\x00\x04\x01d\x0b\x01\x00\x02\xff\xff']
[[1193806329.95552, 108, 108], 'a\x88\xdf.\x1b>\x14\x9c\x17\x08\x00\x01\x00+\x18\x08\x0e\x01d\x01\x00\x04\x01d\x0b\x01\x00\x02\x00\x04\x01d\x0b\x01\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff']
[[1193806329.95552, 112, 112], 'a\x88\xdf.\x1b>\x14\x9c\x17\x08\x00\x01\x00+\x18\x08\x0e\x01d\x05\x01\x04\x01d\x0b\x01\x00\x02\x00\x04\x01d\x0b\x01\x00\x02\x00\x00\xe74\xf2\xbb\xa7\xe4\xd1\x00\xb2#\xef\\#\xa9\x9c\x00\xff\xff']
[[1193806329.95552, 112, 112], 'a\x88\xdf.\x1b>\x14\x9c\x17\x08\x04\x01\x00+\x18\x08\x0e\x01d\x05\x01\x04\x01d\x0b\x01\x00\x02\x00\x04\x01d\x0b\x01\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff']

rmspeers, 2018-07-28:
Indeed, it looks like that PCAP sample (it's been in there forever) may have been captured without the FCSs being corrected.
A bit ago I tweaked the code to recorrect FCSs before writing to PCAP as often devices don't give you the actual FCS, as they check it in firmware and replace it with other data bits (RSSI, LQI, FCS booleans, etc).

@NickEONeill  when you capture with the RZUSB stick, which firmware version do you have loaded, and what do you get when you view the output PCAP file from say zbdump in Wireshark?

NickEONeill, 2018-07-31:
I've currently got version 5 of the firmware on the stick, I've just tested running zbdump and viewing the output in wireshark and it still shows with 'Bad FCS' on a few of the rows. I also have one row that is showing as 'Fragment or Frak [Malformed Packet]'

hgbstu, 2018-11-27:
@rmspeers I'm sniffing ZigBee communication with the RZRAVEN and FW Killerbee006 and I'm also getting sometimes Bad FCS and Malformed packets, see the attached log starting from Line 74. Do you have an idea how I could debug this further?

[zb_c25_20181127-151451.zip](https://github.com/riverloopsec/killerbee/files/2620738/zb_c25_20181127-151451.zip)

fhred, 2018-12-05:
@rmspeers Same problem here.  
Every IEEE Data frame states **Bad FCS**

rmspeers, 2018-12-05:
I looked at @hgbstu 's log, and only a small percentage are reporting bad FCS. This is likely as expected -- RF has error rates in transmission, resulting in some frames being corrupt (or their FCS being corrupt), and thus failing the FCS check that Wireshark is doing for you. The attached image shows a the small set which fail FCS checking. Does why these occur in this case make sense?
[image]

@fhred  I would not expect every frame to corrupted (aka have a bad FCS or not parse properly) unless you are capturing in a congested RF environment (e.g., lots of WiFi, etc). What is your antenna setup/environment/etc? Can you share the PCAP (feel free to email if you can't post here).

rmspeers, 2018-12-05:
@NickEONeill  your 7/31 situation sounds like @hgbstu 's that I answered above, does that fit your situation?

fhred, 2018-12-06:
@rmspeers that's what my first thought was, so I took the devices I'm testing -- Philips Hue Hub and Lightbulb -- away from the very noisy university environment (more then 15 WiFi networks) to my home network (only a handful WiFi networks from neighbors). The results are the same (291 packets, where 221 have a bad FCS). I would greatly appreciate, if you could have a look at this `.pcap` file, as I have no idea what the problem is.

[zbwireshark_2018-12-06-12h50.zip](https://github.com/riverloopsec/killerbee/files/2652967/zbwireshark_2018-12-06-12h50.zip)

rmspeers, 2018-12-15:
Replied to @fhred  in email -- issue could be that the ApiMote is set for an external antenna and none is attached, causing an ~72% of frames to be received corrupted.

fhred, 2019-02-01:
@maxjung96 Yea, that didn't solve the problem for me either. It still persists on my side of things. 
No you don't have to adjust any settings, in theory the Apimote should work that way.

taylorcenters, 2021-07-01:
We believe the apimote reliability has been improved with the latest PRs. Please check it out and reopen if this issue persists.

closed 2021-07-01 by taylorcenters