riverloopsec/killerbee #140: Discontinuation of RZUSB Stick

joswr1ght, 2018-07-31:
A friend sent me a link to the discontinuation notice for the RZUSB stick:

https://www.mouser.com/ProductDetail/Microchip-Technology-Atmel/ATAVRRZUSBSTICK?qs=txwRF2uft8wSedIxfUdhAA%3D%3D

Thoughts on alternative, inexpensive hardware?

rmspeers, 2018-07-31:
I'm thinking the CC2531 dev boards are probably good to be using more often now -- similar form factor, price, and such. We have basic support for those in now.

rmspeers, 2018-11-07:
We also now have support for many of the Ember sticks. Closing this but feel free to reopen as needed.

closed 2018-11-07 by rmspeers

yesimxev, 2019-02-26:
@rmspeers - just to make sure I would select the correct one, Cortet EM358 USB Stick would work? I wanted to buy RZUSB too and didn't find anything under 100$

rmspeers, 2019-02-27:
I haven't had an opportunity to evaluate that specific device so I can't confirm that. I have had good success with the CC2531 EMK for RX mode operations - and with effort one should be able to add TX to that as well.

yesimxev, 2019-02-27:
Which ember sticks/chips do you support recently? Just looking for one which would work for TX