riverloopsec/killerbee #142: IMPORTANT: Can't flash the RZ RAVEN USB Stick!

mrnice1, 2018-08-27:
Hello guys,
I hope you can help me since I'm in a hurry because of my studies. 
I'm trying for several hours to flash the RZ RAVEN USB Stick with the ATAVRDRAGON, but it simply doesn't work. I also tried another RZ Usb stick, and it doesn't work either. 

First of all I'm using Ubuntu 16.04 on a MacBook Air (Early 2015) and I use all requisites written in the Readme.

Here is what I get, when I enter the following commands:

> cd killerbee/firmware
> $ sudo avrdude -P usb -c dragon_jtag -p usb1287 -B 10 -U flash:w:kb-rzusbstick-006.hex 

I get this:

> avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_JTAG_ID
avrdude: retrying with external reset applied
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_JTAG_ID
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
avrdude done.  Thank you.

When I enter 

> avrdude -P usb -c dragon_jtag -p usb1287 -B 10 -U flash:w:kb-rzusbstick-006.hex -vvvv

I get this:

> $ sudo avrdude -P usb -c dragon_jtag -p usb1287 -B 10 -U flash:w:kb-rzusbstick-006.hex -vvvv

avrdude: Version 6.2
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/ahmad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : dragon_jtag
         Setting bit clk period        : 10.0
avrdude: jtagmkII_dragon_open()
avrdude: usbdev_open(): Found AVRDRAGON, serno: 00A20007067C
avrdude: jtagmkII_getsync()
avrdude: jtagmkII_getsync(): Sending sign-on command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Sent: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97] 
avrdude: jtagmkII_recv():

avrdude: jtagmkII_recv(): Got message seqno 0 (command_sequence == 0)
avrdude: Recv: . [86] . [01] . [ff] . [0b] . [06] . [01] . [ff] . [0b] . [06] . [07] . [00] . [a2] . [00] . [07] . [06] | [7c] A [41] V [56] R [52] D [44] R [52] A [41] G [47] O [4f] N [4e] . [00] 

Raw message:
0x86 0x01 0xff 0x0b 0x06 0x01 0xff 0x0b 0x06 0x07 0x00 0xa2 0x00 0x07 0x06 0x7c
0x41 0x56 0x52 0x44 0x52 0x41 0x47 0x4f 0x4e 0x00 
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              6.11
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              6.11
  hardware version:              7
Serial number:                   00:a2:00:07:06:7c
Device ID:                       AVRDRAGON
avrdude: jtagmkII_getsync(): Using a 298-byte device descriptor
[... 231 lines cut ...]

Raw message:
0xa9 
Illegal JTAG ID

avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_JTAG_ID
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtagmkII_close()
avrdude: jtagmkII_close(): Sending sign-off command: 
avrdude: jtagmkII_send(): sending 1 bytes
avrdude: Sent: . [1b] . [0e] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] w [77] . [b6] 
avrdude: jtagmkII_recv():

avrdude: jtagmkII_recv(): Got message seqno 14 (command_sequence == 14)
avrdude: Recv: . [80] 

Raw message:
0x80 
OK

avrdude done.  Thank you.

rmspeers, 2018-08-27:
I presume you've already read it, but https://github.com/riverloopsec/killerbee/blob/master/firmware/README.md details a few different ways depending on what hardware. For using the AVR Dragon, you could either flash the USB DFU updater that @AdamLaurie wrote or flash the firmware directly. CC @jrussell88 who also got it working with the default DFU USB bootloader from Windows.

rmspeers, 2018-08-27:
Also are you sure you have excellent connections throughout the communication on your header? E.g. hold or solder in place for good contact?

mrnice1, 2018-08-28:
Rello Ryan,

thank you for the fast response. I already tried flashing the DFU Bootloader, but I still get the same error. And I'm very sure, that the header connection is good since I'm holding it in an angle.

mrnice1, 2018-08-28:
I finally found a solution for my issue: The JTAG cable wasn't working, so I tried the AVR Wireless Method on a Virtual Machine with Windows 7 x86.
At first it didn't work. After the first flashing process I got an error, but the RZ RAVEN turned Amber. Anyway zbid still showed me the RZUSBSTICK and not the KILLERB006. 
The solution is: You have to leave the RZUSBSTICK inserted and start the Upgrade / Flash progress again. Then the RZUSBSTICK turns green / yellow and the Flashing was successful. 

Sometimes it shows you an error, that there is no usb stick found. Therefore you have to close the virtual machine and save the actual.
After that you have to delete it from the usb filter and reinsert it. Keep in mind that the RZUSBSTICK is still inserted. Then you start the virtual machine again and start the upgrade progress. After that it should work.

You may add it in the description :)

Best regards

closed 2021-06-22 by taylorcenters