riverloopsec/killerbee #143: zbreplay not working

mrnice1, 2018-08-29:
Hello guys,
it's me again.

I'm using an RZUSBSTICK with the latest firmware on it. I also use Ubuntu 16.04
zbreplay won't work, I always get this error:

 sudo zbreplay -r huedumpreplay.pcap --channel 15
Warning: You are using pyUSB 1.x, support is in beta.
Traceback (most recent call last):
  File "/usr/local/bin/zbreplay", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbreplay')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 719, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1511, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbreplay", line 72, in <module>
    
TypeError: is_valid_channel() takes exactly 3 arguments (2 given)

How can I solve it.

riverloopsec, 2018-08-30:
This looks like it may be a bug on our end and I will investigate.

ichintu, 2018-10-11:
I am having this issue too. Affects zbjammer too. If anyone has a fix please let me know.

Nonnie8665, 2018-11-07:
So this may be a crappy fix, but I got it to work... I went to the zbreplay.py in the killerbee/tools folder on line 72 and commented out "if not kb.is_valid_channel(args.channel):" . Afterwards, I replaced it with the some code from kbutils.py and re-ran setup.py.  Hopefully this works for everyone till the bug gets resolved:

signal.signal(signal.SIGINT, interrupt)
#if not kb.is_valid_channel(args.channel): # This blows up zbreplay
if not (args.channel >= 1 and args.channel <= 26): # This was taken directly from kbutils.py is_valid_channel() function
     print >>sys.stderr, "ERROR: Must specify a valid IEEE 802.15.4 channel for the selected device."
     kb.close()
     sys.exit(1)
kb.set_channel(args.channel)

rmspeers, 2018-11-07:
Agreed, this change is in the branch and is waiting to merge for the next release which I hope will be soon. If not I will do a dot release into master to resolve this issue.

Nonnie8665, 2018-11-08:
Awesome!  Thanks for the quick response and bugfix turnaround.  I just applied your changes :-)