riverloopsec/killerbee #145: Cannot Flash Api-Mote

ichintu, 2018-10-16:
Hello,

I have purchased a new Api-Mote from Attify stores. When I attempt to flash the firmware it fails:

> root@kali:/opt/killerbee/firmware# ./flash_apimote.sh
> Flashing apimotev4_gf.hex to the connected USB serial device.
> MSP430 Bootstrap Loader Version: 1.39-goodfet-8
> Invoking BSL...
> Transmit default password ...
> Current bootstrap loader version: 2.13 (Device ID: f26f)
> Checking for info flash...  None.
> Look at contrib/infos/README.txt for better performance.
> Mass Erase...
> Transmit default password ...
> Invoking BSL...
> Transmit default password ...
> Current bootstrap loader version: 2.13 (Device ID: f26f)
> Changing baudrate to 38400 ...
> MSP430 Bootstrap Loader Version: 1.39-goodfet-8
> Invoking BSL...
> Transmit default password ...
> Traceback (most recent call last):
>   File "./goodfet.bsl", line 1982, in <module>
>     main(0);
>   File "./goodfet.bsl", line 1882, in main
>     bsl.actionStartBSL()
>   File "./goodfet.bsl", line 1164, in actionStartBSL
>     self.txPasswd(self.passwd)                  #transmit password
>   File "./goodfet.bsl", line 1134, in txPasswd
>     wait=wait)           #if wait is 1, try to sync forever
>   File "./goodfet.bsl", line 800, in bslTxRx
>     rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
>   File "./goodfet.bsl", line 479, in comTxRx
>     rxHeader, rxNum = self.comRxHeader()        #receive header
>   File "./goodfet.bsl", line 385, in comRxHeader
>     if not hdr: raise BSLException("Timeout")
> __main__.BSLException: Timeout

I have tried this several times across several laptops running Ubuntu / Kali etc with no luck. Please help.

cn0xroot, 2018-10-23:
Have the same issues.
Did you fixed it now?

rmspeers, 2018-10-23:
It worked when the user flashed it when they didn't provide the `--speed` argument, however the devices shouldn't need to be flashed. Please ensure you are using it on a supported kernel USB version, e.g., test on a bare-metal box on Ubuntu 12.04 and see if it works out-of-the-box there.

cn0xroot, 2018-10-24:
First of all,thanks for your quickly replay.
Api-Mot it worked on ubuntu 16.04,and it flashed success with your tips.
```
lsusb
Bus 003 Device 007: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
```

```
sudo ./goodfet.bsl -e -p apimotev4_gf.hex
Board not specified.  Defaulting to goodfet41.
Press Ctrl+C to cancel, or Enter to continue.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Program ...
7874 bytes programmed.
```

ichintu, 2018-10-25:
@cn0xroot have you tried running zbwireshark or zbstumbler after you re flashed and is it working or are you seeing serial timeouts?

cn0xroot, 2018-11-06:
@ichintu  maybe you need run it with sudo :

```
./zbstumbler 
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Warning: waiting for serial read timed out (most likely).
^C
1 packets transmitted, 0 responses.

sudo ./zbstumbler

zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
^C
17 packets transmitted, 0 responses.
```

it work fine with root,but i didn't see any data packet when zbwireshark start (try channel 11 、14 15)...
TI cc2531 dongle can capture packet on channel 11 and 15 @rmspeers  what error with api-mote?

rmspeers, 2019-01-24:
It sounds like this is something we can now mark resolved, with the following notes:
- you must have permission to talk to the serial device, thus update your OS's groups on that device or use sudo (not recommended)
- If you have trouble flashing, remove the `--speed` argument from the flashing command and try

This will be closed but can be reopened if needed.

closed 2019-01-24 by rmspeers