riverloopsec/killerbee #146: Using european frequency (868 MHz)

whoot, 2018-11-01:
Hello,
I´m using Killerbee with Open Sniffer from SEWIO in v3 (firmware version 0.9.0).
In Europe ZigBee is using 868 MHz as a frequency - and I think it is using just one channel (0)
zbstumbler is just cycling through channels 11 to 26, which are used in 2.4 GHz. 

I know that I can change the code to also search on channel 0, but how/where can I change the frequency?

whoot, 2018-11-01:
Alright, i changed the code and now zbstumbler supports 868 MHz frequency 👍 
I will try to add the support for other frequencies too and then make a push request

joswr1ght, 2018-11-01:
That would be fantastic. Thank you!

rmspeers, 2018-11-01:
Indeed, thanks! The method should be possible via the pages argument, as that is how other firmware has support for these bands.

I have merged your other PR cleaning up the CLI for zbstumbler, and will try to get the branch it was PRed into merged down to master when possible.

whoot, 2018-11-01:
Yeah, you should definetly pull that into master, because the device and the channel are not working there, too. 
I will push my fixed version tomorrow.
I dont know if it would be useful to add a new parameter (like --zone europe/china/us or something like that) in order to be able to trim down the channels. On the other hand you can specify them with the channel parameter. Any ideas?

whoot, 2018-11-01:
Pull request created.
zbstumbler just cycles through all channels now.

closed 2018-11-01 by whoot