riverloopsec/killerbee #152: Device identifier (USB devices) confusion

fhred, 2018-11-05:
Hi everyone,

I've noticed some unexpected (for me at least) behavior using the `zbwireshark` tool. As  `zbwireshark` has the `-i` flag I was trying to use it's options and getting familiar with the code base. So I tried to figure out the rather cryptic information from  `zbwireshark -h`: 
```
-i DEVSTRING, --iface DEVSTRING, --dev DEVSTRING
                      Device to use for sniffing
```
And quickly found, that `devstring` was used in the constructor of the `Killerbee` class as the `device` value. Reading that code snipped it gave me the information:  `@param device:  Device identifier, either USB vendor:product, serial device node, or IP address`.  
So I ran `lsusb` in my bash console which gave me `Bus 001 Device 007: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)` and therefore tried `zbwireshark -i 0403:6015 -c 25`, as `0403` and `6015` are the product and the vendor ID.  
That didn't work so I digged deeper into the code and found, that the constructor divides into three options for the `device` parameter -- as mentioned in it's documentation (that code snipped is rather confusing IMHO) -- and calls `kbutils.search_usb(...)` in the case I was struggling with.  
Looking at `search_usb` I got this information:
```
Takes either None, specifying that any USB device in the
global vendor and product lists are acceptable, or takes
a string that identifies a device in the format
<BusNumber>:<DeviceNumber>, and returns the pyUSB objects
for bus and device that correspond to the identifier string.
```
which describes precisely what the function actually does. So I've tried `zbwireshark -i 001:007 -c 25` which didn't work as well.  
After some more digging I realized, that the Api-Mote strings (in `kbutils`)
```
FTDI_X_USB_VEND_ID  = 0x0403
FTDI_X_USB_PROD_ID  = 0x6015
 ```
are not contained in the `usbVendorList` and `usbProductList` which makes it impossible to find an Api-Mote using the function. Simply putting them in just 'shifts' the error to another place, namely to: `KillerBee doesn't know how to interact with USB device`.  
As of that I had a closer look at the constructor again and noticed, that the devices that can be added using the `<BusNumber>:<DeviceNumber>` format are disjoined from the ones that can be added using a `/dev/tyy*` format. Is there a reason for that, if yes why is it not documented anywhere (at least I didn't find it)?

For completness:
* `zbwireshark -c 25`
* `zbwireshark -i /dev/ttyUSB0`
are working perfectly.

**tl;dr**
So I've found two things (if I'm not mistaken -- so please correct me if I just misunderstood something):
* The two documentations (`Killerbee` constructor and `search_usb`) are contradicting each other.
* The Api-Mote and others cannot be added unsing the `<BusNumber>:<DeviceNumber>` format, others can only be added that way.

rmspeers, 2018-11-05:
You are correct that ApiMote will not work with `<BusNumber>:<DeviceNumber>`, as it's firmware speaks USB serial, and thus expects a `/dev/ttyUSB0` type definition. The USB bus/device definition is used for items that speak using pyusb.

Sounds like we should resolve the documentation strings, however!

rmspeers, 2018-11-05:
@fhred  please change to the release/2.7.1 branch and see if that now helps. Please provide feedback from installing with it, etc. https://github.com/riverloopsec/killerbee/pull/153.

fhred, 2018-11-06:
@rmspeers install from release/2.7.1 went flawlessly and I cannot find any error after some minutes of testing.  
The documentations are no longer contradicting each other, that's fixed.  

I wondered if I am the only/first person to stumble upon this, I guess it would help to write something along the lines of: 
> Depending on whether the device's firmware speaks pyusb or USB serial either the `<BusNumber>:<DeviceNumber>` or the serial device path (e.g. `/dev/ttyUSB0`) option will work.

to prevent confusion for noobies like me.

rmspeers, 2018-11-07:
This issue will be closed when the 2.7.1 release is merged to master.

PR #153 (2019-06-04): Updated for 2.7.1 packaging.
Resolves #152 
Resolves #149
commit bee32317 (2019-06-04): Updated for 2.7.1 packaging. Resolved #152.
closed 2019-06-04 by rmspeers