riverloopsec/killerbee #154: zbfind not working properly with Api-Mote

wveilleux, 2018-11-16:
Hi, I'm new to killerbee and when I start zbfind, it never show any signal strenght (zbwireshark is working very well). Here is the output I got:

sudo zbfind -c 15
[sudo] password for bs5360:
# DEBUG Clearing overflow
# DEBUG Clearing overflow
# DEBUG Clearing overflow
# DEBUG Clearing overflow
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbfind", line 1039, in run
AttributeError: type object 'datetime.datetime' has no attribute 'datetime'

wveilleux, 2018-11-21:
No clue anyone ? 
I'm not a coder but it seems it come from the import of datetime using datetime.datetime in the zbfind code, I did a few googling https://www.google.com/search?q=killerbee+zbfind+AttributeError+type+object+datetime.datetime+has+no+attribute+datetime&ie=utf-8&oe=utf-8&client=firefox-b-ab but I can't figure it out.
Thanks

wveilleux, 2018-11-21:
I found a workaround by changing the "datetime.datetime" for "datatime" like this:

diff zbfind.orig zbfind
1023c1023
<                     details[0][1] = "Last seen: " + str(datetime.datetime.now())
---
>                     details[0][1] = "Last seen: " + str(datetime.now())
1039c1039
<                 nowstr = str(datetime.datetime.now())
---
>                 nowstr = str(datetime.now())

with this "patch", I can use zbfind properly even in passive or active mode.

wveilleux, 2018-11-21:
python -V
Python 2.7.15rc1

on Ubuntu Server 18.04 (bionic)

taylorcenters, 2021-06-22:
closed - please try again on most recent versions and reopen with updated output if still an issue

closed 2021-06-22 by taylorcenters