riverloopsec/killerbee #155: Api-Mote Recognition Error

SQLisLOVE, 2018-12-15:
Hey, so I have an Api-Mote board that cannot be recognized after running zbid. Every time I run the command it just fells me to go to GoodFet to look for missing flash issue. I have reprogrammed the board several times but it still cant be recognized. It was working fine 2 days ago but now is incapable of running commands. Anyway, if you have any suggestions on what i could do plz leave them down below. Thank you for your time!

rmspeers, 2018-12-15:
This is likely due to how the board was reprogrammed, although the initial issue can happen sometimes due to how the GoodFET framework that we use to communicate to the chip works. I would suggest also checking to see if you had an OS upgrade. What OS are you running on now, and was it updated recently at all?

SQLisLOVE, 2018-12-15:
I believe I am running Ubuntu 16.04. I have reprogrammed the board before on this same OS so I don't understand what's different now. And no, I have not updated recently.

rmspeers, 2018-12-15:
Ok, interesting to hear it worked before. Ubuntu 16.04 is not something I expect ApiMote v4beta to work reliably on - likely due to the FTDI chip and changes to the Ubuntu USB handling between 12.04 and 16.04. I personally use 12.04 for more reliable communication but I'm also sure a 16.04 user could tweak the FTDI chip's settings or linux side settings to get it working reliably.

SQLisLOVE, 2018-12-23:
What setting should i look at and what would you recommend changing it to? I'm only a first year research student and am a bit of a novice. Thank you for your help and Merry Christmas!

taylorcenters, 2021-07-01:
long inactive - closing. Please update to latest version and open a new issue if you still have problems. Thank you.

closed 2021-07-01 by taylorcenters